frankarobotics/libfranka #8: Update controller to work with develop branch?

tverbele, 2018-01-08:
Is there a way for us to update the controller so that we can experiment with the latest develop branch of libfranka?

ch1bo, 2018-01-08:
As the changelog on the develop branch indicates, it will require the upcoming system version (probably 1.1.0). I cannot give you an exact time point when this update will be available, but we are currently testing it and plan to release it soon. The update will be available for download from support.franka.de (frankaemika.github.io)

fwalch, 2018-01-30:
@tverbele Panda system version 1.1.1 is now available from https://frankaemika.github.io. `franka_ros` and `libfranka` have been released as well and include the changes from the `develop` branch.

closed 2018-01-30 by fwalch