frankarobotics/libfranka #9: controller_torque_discontinuity settings

smithjoshua001, 2018-02-05:
Hello,
I am wondering whether there is a way to change the behaviour of the torque_discontinuity error. Mostly to the point of allowing larger instant changes in torque?
-Josh

fwalch, 2018-02-06:
This threshold is enforced by Control and cannot be changed by the user. What you can do is perform rate limiting (see `tau_d_saturated` in `examples/joint_impedance_control.cpp`).

smithjoshua001, 2018-02-13:
Ok thanks, also I was wondering will there be support for sending raw torques to the robot i.e. without the gravity and friction compensation added by the robot? or is that controlled by the control too?

fwalch, 2018-08-09:
I just realized I never answered you again, sorry for that! I'm not sure if this will still be useful after all this time, but: as the robot will add gravity, you could try [calculating and subtracting it](https://frankaemika.github.io/libfranka/classfranka_1_1Model.html#af7c9dd1bafa5404c7a208d49778c3712) from the torques you send in your control loop and then add your own values. Other than that, there isn't a possibility to send raw torques.

We also added more documentation about the [control parameters](https://frankaemika.github.io/docs/control_parameters.html) and [how the robot is controlled by libfranka](https://frankaemika.github.io/docs/libfranka.html#under-the-hood).

closed 2018-09-04 by fwalch