frankarobotics/libfranka #13: Grasp with force not working after 0.3.0 upgrade

tverbele, 2018-03-01:
When I grasp an object with force using libfranka, this no longer works after the 0.3.0 upgrade. 

For example running the `grasp_object` example, it only applies force for a fraction of time and then just stops. However, when I use the close with force from the Pilot it does work. Has this anything to do with the new epsilon arguments in the grasp method?

fwalch, 2018-03-02:
The example should work. Can you post how you invoke the `grasp_object` example on the command line and its output?

tverbele, 2018-03-05:
I invoke `./grasp_object $ROBOT_IP 1 0`

I wait until the homing is done and then hold a something between the gripper to close. The output is `Failed to grasp object.`

fwalch, 2018-03-05:
As you suspected, this is indeed caused by the new epsilon handling. The `grasp` method will now only consider an object grasped if the actual grasping width `d` relates to the commanded width `width` (in your case, `width=0`) as `width - epsilon_inner < d < width + epsilon_outer` ([`franka::Gripper::grasp docu`](https://frankaemika.github.io/libfranka/classfranka_1_1Gripper.html#a19b711cc7eb4cb560d1c52f0864fdc0d)). To allow the grasping to succeed in your case, you can increase `epsilon_outer`.

We should have pointed out this change in behavior in the changelog, sorry about that.

tverbele, 2018-03-05:
Thanks, that explains. Maybe adapt the grasp example code to allow it to grasp an object of any width with the default parameters?

fwalch, 2018-03-05:
You can already give an arbitrary object width to the example via the command line parameter. This is to verify that you grasped the expected object (or indeed any object at all), so we don't want to change that for the example.

tverbele, 2018-03-06:
Ok thanks, this issue can be closed then!

closed 2018-03-06 by tverbele