frankarobotics/libfranka #16: example generate_cartesian_pose_motion fails

MassaRoddel, 2018-03-27:
I want generate a cartesian motion. Therefore, I took the example as a starting point.
Unfortunally, the example generate_cartesian_pose_motion fails with error:
libfranka: Attempt to set invalid transformation in motion generator. Has to be column major!

What is the problem?

PS: today I encountered the timing problem again and the fix was to reboot the robot controller.

fwalch, 2018-03-27:
This problem can occur if the end effector configuration set in the Admin interface is not precise enough, causing the resulting transformation matrix to not be within the thresholds in libfranka. Can you check this? For example, for the Hand, the values for the rotation matrix should be "0.7071, -0.7071, ...".

About the second problem: thanks for the hint, we are aware of the problem and are working on it.

MassaRoddel, 2018-03-27:
Changing the digits from 0.707 to 0.7071 fixed it.
Thank you

fwalch, 2018-03-27:
You're welcome!

closed 2018-03-27 by fwalch