frankarobotics/libfranka #17: API for opening brake?

MinhDucLe2412, 2018-04-10:
Is there any plan to release API to open and close brake?

edmundoferreira, 2018-04-10:
That would be a nice feature indeed.
I also search the API but couldn't find any way to do that from the libfranka C++ side.

@MinhDucLe2412 are you also manually going to webinterface to open/close or reset the brakes on errors? Or are you doing something more fancy?

dudalu, 2018-04-10:
Unfortunately, we have no plans to support opening and closing brakes from libfranka. 
This has to be done through the web interface. 

However, you do not have to toggle the brakes to recover from errors. Try toggling the "external activation device" (the one connected to X4), or calling ``robot.automaticErrorRecovery()``.

closed 2018-04-10 by dudalu