frankarobotics/libfranka #19: Changing orientation of End Effector fails

MinhDucLe2412, 2018-06-12:
I'm trying to change the orientation of EE by writng new values to O_T_EE_d field but somehow i got this error: 

"libfranka: Attempt to set invalid Transformation in motiongenerator. Has to be column major "

I'm using Hand with the following configurations:

Mass 0.73 kg
Flange to Center : [-0.01 0 0.03]
Inertia Tensor : [0.001 0 0 0  0.0025 0 0 0 0.0017]
Transformation Matrix [0.7071 -0.7071 0 0 0.7071 0.7071 0 0 0 0 1 0 0 0 0.1034 1]

Any help would be appreciated

sgabl, 2018-06-19:
Could you please provide the matrix you're trying to set?

MinhDucLe2412, 2018-06-19:
Do you mean this?

"O_T_EE_d": [0.996423,0.0434016,0.0723757,0,0.0459562,-0.998355,-0.0340119,0,0.0707818,0.0372171,-0.996797,0,0.498007,-0.0139371,0.111537,1]

sgabl, 2018-06-19:
This pose should be valid. Could you provide the relevant part of the control loop?

closed 2018-08-09 by fwalch