frankarobotics/libfranka #20: communication_constraints_violation after update

MassaRoddel, 2018-09-18:
After updating firmware to 1.3.2 and libfranka to 0.5.0 I always get communication_constraints_violation errors. The example communication_test fails too.
The ping test results are no different from before the update.
My code always fails with the same value of 0.78 now. Do we need more cpu power for version 0.5.0 than 0.3.0? Could this be the problem?

jrmout, 2018-09-18:
There should be no difference. Maybe you forgot to compile in Release mode ( cmake -.DCMAKE_BUILD_TYPE=Release .. )?

jrmout, 2018-09-19:
If you don't specify the CMAKE_BUILD_TYPE variable then it will be empty and no additional flags will be set.

MassaRoddel, 2018-09-19:
Now communication works but the code to move to cartesian coordinates does not. The robot still moves but much less distance and without errors.
ie  robot should move 100mm but it moves only 4mm.
Any ideas what the problem could be?

jrmout, 2018-09-19:
The last release added lowpass filters on libfranka side. As you edited, deactivating those should help.

closed 2018-09-19 by jrmout