frankarobotics/libfranka #23: self_collision_avoidance_violation with collision free motion plans

5yler, 2018-09-27:
We are sending the robot collision-free trajectories which perform as expected in simulation. However, when running on the physical robot, it will often stop with a `self_collision_avoidance_violation` error. We have tried replacing the mesh collision geometry in the `franka_description` URDF with cylinders/boxes and specifically increasing the cylinders around links 1 and 6 to address the problem, but the robot will still stop - see below screenshots with overlaid collision geometry in an example configuration.

[image]

[image]

Is there a way to disable the self collision avoidance or change parameters such that this error would not be triggered during feasible trajectories? Increasing the collision geometry even further limits reachability in our workspace.

kribr, 2018-10-17:
We are encountering the same problem as well when panda link 2 and 6 move close to one another. We have found the collision error to be reproducible, but have not found a solution yet.

jonasTUM, 2019-07-16:
Hello everybody, I just faced the same issue. Could you let me know how you proceeded with that problem?

fwalch, 2019-07-22:
In the develop branch on `franka_ros`, we updated the collision volumes to match the ones we use internally. This has not been released yet, but we'll try to do that soon. In the meantime, you can try to use the develop branch. I'll therefore close this as a duplicate of https://github.com/frankaemika/franka_ros/issues/39

closed 2019-07-22 by fwalch