frankarobotics/libfranka #24: Are there plans to add inverse kinematics to the model library?

Tomakko, 2018-10-24:
Or can i obtain the inverse kinematics in any other way?

fwalch, 2018-10-24:
There are no plans at the moment. You can e.g. use the library which is also used by ROS, [orocos_kdl](https://github.com/orocos/orocos_kinematics_dynamics).

fwalch, 2018-10-24:
@gavanderhoorn Thanks for this suggestion!

> Also comes as a stand-alone version (ie: no ROS)

It seems that even the `trac_ik_lib` has a lot of catkin dependencies. Am I missing something?

miguelprada, 2018-10-24:
A small warning on `trac_ik`. It can certainly solve some IK requests where KDL would fail but it can also, on certain conditions, return solutions that require configuration changes with respect to the provided seed configuration. Just be careful of checking what it returns if your use case requires _continuity_ in the joint configuration.

gavanderhoorn, 2018-10-24:
> > Also comes as a stand-alone version (ie: no ROS)
> 
> It seems that even the `trac_ik_lib` has a lot of catkin dependencies. Am I missing something?

hm, either I'm misremembering, or something changed since I last looked at it.

gavanderhoorn, 2018-10-24:
@miguelprada: did that also happen when you configured the `Distance` solve type?

gavanderhoorn, 2018-10-24:
@fwalch: just curious. Why do you guys not provide an FK solver for your robots?

miguelprada, 2018-10-24:
> @miguelprada: did that also happen when you configured the Distance solve type?
>
> Default is Speed and that can do what you describe. With Distance this has not happened for me (but I haven't run 1e9 queries, so this may be lucky sampling).

To be honest, I haven't played with it a lot myself either. A colleague, @asierfernandez, was trying to use it to improve the solver performance for a dense trajectory of cartesian end-effector poses, where we definitely do not want configuration changes between one sample and the next, and he had some issues with this. I'm not sure how this effort ended up.

gavanderhoorn, 2018-10-24:
this is getting off-topic here, but: check the configuration, as I wrote, `Speed` is the default and gets used if you don't change anything. Configuration changes happen all the time. `Distance` should sort that out.

SaraHamdan, 2020-09-23:
> You might want to consider [trac_ik](http://wiki.ros.org/trac_ik) instead.
> 
> Also comes as a stand-alone version (ie: no ROS) and is 'better' than KDL in most cases.

@gavanderhoorn would you please share the link of the stand-alone trac_ik ? (all I found is the ROS one)

AndreasKuhner, 2024-10-30:
If there is anything further to discuss, please re-open the issue. Thanks to everyone :smile:

closed 2024-10-30 by AndreasKuhner