frankarobotics/libfranka #27: Robot moves to worng cartesian rotation angles

MassaRoddel, 2018-11-12:
Since the introduction of rate limiter and lowpass filter the robot often moves to wrong cartesian rotation angles (about 50% chance). It works again if rate limiter and lowpass filter are disabled and setFilters() gets used.
The wrong angles seem to be always the same for the same destination angles.
The x,y,z coordinates are always correct.

Tomakko, 2018-11-21:
I can verify the issue. 
MasseRoddel, can you tell me how to get it working? 
I am doing `robot.control(cartesian_pose_control_callback,franka::ControllerMode::kJointImpedance, false, 1000.0);`
and
`robot.setFilters(1000,1000,1000,1000,1000);`
but the issue is still the same.

MassaRoddel, 2018-11-22:
I (have to) use robot.setFilters(5, 5, 5, 5, 5) to avoid acceleration/velocity discontinuity errors.

My ping test results are like:

60000 packets transmitted, 60000 received, 0% packet loss, time 60048ms
rtt min/avg/max/mdev = 0.144/0.217/0.365/0.010 ms

but still the success rate goes to low values.

Control command success rate of 9944 samples: 
Max: 1.00
Avg: 0.96
Min: 0.76

MassaRoddel, 2018-11-22:
Does that mean that executing the control loop in a thread different from main thread makes it worse?

tf-janr, 2018-11-22:
No, in this case we just included a library into the main.

Tomakko, 2018-11-22:
Thanks, @MassaRoddel, setting your filter coefficients seems to fixed it indeed.

Tomakko, 2018-11-22:
@tf-janr , how does the differences in reliability look like? You experience more of the errors your described when placing the cartesian pose callback function inside main?

tf-janr, 2018-11-22:
It works more reliable direct in main. If you build the loop in separate class and include this library to main the rotation error occurs more often. 
The calculation of rotation works fine because O_T_EE_c is always correct. But when we get the orientation error the O_T_EE_d shows unexpected behavior.

tf-janr, 2018-11-23:
We got exactly the same answer

fwalch, 2019-01-22:
We finally have a system update available which should fix this issue. You can download it [here](https://s3.eu-central-1.amazonaws.com/franka-updates/update-2.1.3.img). This is compatible with libfranka 0.5.0.

MassaRoddel, 2019-01-28:
Seems not to be fixed but it happens less often now.

pantor, 2019-01-28:
I can confirm that the system update does not fix our orientation problems.

fwalch, 2019-01-28:
Thanks for the feedback. We also received reports of this through support and are investigating.

MassaRoddel, 2019-07-05:
Is this problem fixed by now or is there a workaround?

fwalch, 2019-07-05:
With system update 2.1.4, this should be fixed. You can download it on https://frankaemika.github.io/
Let me know if you are still experiencing issues.

sgabl, 2019-07-22:
I'm closing this issue. If the problem still persists, please contact support.

closed 2019-07-22 by sgabl