frankarobotics/libfranka #28: Blocking Gripper API can't be interrupted

flederbaysian, 2018-12-24:
This is related to #18, which shouldn't have been closed, IMO.

The [Gripper API](https://frankaemika.github.io/libfranka/classfranka_1_1Gripper.html) offers blocking operations for moving and grasping. 

For my application, I would really like non-blocking operations for moving the gripper. i.e I want `move(0, 0.01)` to immediately, and the gripper should close at 1cm a second until issued a different command.

I tried, in a few ways, to emulate that. All failed:
1. Call `move` on one thread, and whenever I want the velocity changed, call `stop` on another. [`stop`](https://frankaemika.github.io/libfranka/classfranka_1_1Gripper.html#a083c93b41d93e75e5d0675de6ccf60fc) is documented as _"Stops a currently running gripper move or grasp"_ but sadly, as pointed out in #18, the method doesn't actually interrupt currently running blocking operations, and instead blocks, too. I don't see how this is useful for anyone.
1. Edit the source of `Gripper.cpp` by commenting out the code that waits for a TCP response for gripper operations ([here](https://github.com/frankaemika/libfranka/blob/5ca631efda3d077e5d9f5f7299d10575d832cfc2/src/gripper.cpp#L19-L30)). This causes a buffer of responses to fill up on the client side, and also makes the gripper hang and require a restart, if operations are called too frequently...
1. Change the TCP timeout on the `Network` object used by `Gripper` to some very small value. This also caused the gripper to hang.

fwalch, 2019-01-08:
Approach (1) should work, but it seems like we have a bug there which we need to fix. Thanks for pointing this out.

flederbaysian, 2019-01-08:
Great. Looking forward to a fix.

I see you work on franka-ros too, so it would be good to verify the fix with the corresponding action servers, too.

fwalch, 2019-01-21:
We are planning to include this fix in the next software release. I hope this will be available by the end of the month, but I cannot give any guarantees.

fwalch, 2019-03-11:
This has been fixed in libfranka 0.6.0/system version 3.0.0. Please contact support@franka.de if you want to receive system update 3.0.0.

closed 2019-03-11 by fwalch