frankarobotics/libfranka #32: Run real-time control with switches

Gathiyo, 2019-02-22:
I'm running Franka Emika Panda arm with switches between control box and workstation. Because of the switch lag is being introduced and robot can not run real-time commands as it drops around 20 packages. 

Considering I can not do anything about the switches how can I run the real-time control at 500Hz instead of 1000Hz so that I can do real-time control with switches. Or is it even possible and if there is any other solution let me know, please.

sgabl, 2019-02-22:
In general it is not possible to decrease the real-time control to 500Hz at the moment. Using certain professional switches would allow to properly control the robot with a 1kHz control loop. However, most consumer switches buffer packets and introduce a high delay / jitter, so 500Hz would probably not even be possible.

I assume you want to use the robot and have at the same time connectivity to your company network / to the Internet? In such a case I'd recommend to buy a second Intel PCI-E network card and directly connect the robot. We generally recommend to only use the robot directly connected to the PC to get the best performance.

Gathiyo, 2019-02-28:
Hello, thank you for your input, we currently have 2 arms connected to the switch which is connected to 3 workstations so that we can control both the arms with any workstation available. One arm s working properly with the switches while the other is not. It keeps giving us communication error.

```
RSL@rsl-panda-experimental1:~/libfranka/build$ ./examples/communication_test 192.168.1.100
WARNING: This example will move the robot! Please make sure to have the user stop button at hand!
Press Enter to continue...

Finished moving to initial joint configuration.

Starting communication test.
.
.
.
#4600 Current success rate: 0.97
libfranka: Move command aborted: motion aborted by reflex! ["communication_constraints_violation"]
control_command_success_rate: -0.7372 packets lost in a row in the last sample: 176

```

For arm 2:
```

RSL@rsl-panda-experimental1:~/libfranka/build$ ./examples/communication_test 192.168.1.200
WARNING: This example will move the robot! Please make sure to have the user stop button at hand!
Press Enter to continue...

Finished moving to initial joint configuration.

Starting communication test.
.
.
.
#9900 Current success rate: 0.96

Finished test, shutting down example
```

Can you please help regrading this issue? 

Note:- Direct communication without the switches does not throw any errors.

sgabl, 2019-07-05:
Hi, I assume that this behaviour is caused by the switch. Interesting would be to check whether using arm 1 on the switch port of arm 2 would lead to an other result for arm 1. However, we recommend to directly connect the control pc to the robot.

sgabl, 2019-07-22:
I'm closing this issue. If the problem still persists, please contact support or reopen the issue.

closed 2019-07-22 by sgabl

josemoreno98, 2024-04-16:
@Gathiyo I have the same problem as you. How did you solve it? Thanks!