frankarobotics/libfranka #38: libfranka example excution return timeout

Junhyuck-Woo, 2019-03-24:
I followed installation of guide line.
(https://frankaemika.github.io/docs/installation.html)

I checked controller and robot network connection.
**--- 172.16.0.2 ping statistics ---
10000 packets transmitted, 10000 received, 0% packet loss, time 9998ms
rtt min/avg/max/mdev = 0.127/0.165/0.264/0.018 ms
--- 192.168.1.1 ping statistics ---
16 packets transmitted, 16 received, 0% packet loss, time 15073ms
rtt min/avg/max/mdev = 3.321/13.904/37.091/9.906 ms**

It works properly. However, when I tried to run "echo_robot_state"  it showed timeout.
**hcr@Panda-PC:~/catkin_ws/src/libfranka/build$ ./examples/echo_robot_state 172.16.0.2
libfranka: Connection timeout**

If you need other information please let me know.
I really need your help to run panda

sgabl, 2019-03-24:
Please open Desk and check `Settings -> System -> Features`. Do you see there the `FCI (Franka Control Interface)` feature? This will be removed after a factory reset. If you don't have this feature file, please contact `support@franka.de`.

Junhyuck-Woo, 2019-03-24:
I really thank for your help~!! I'd like to do double checks.
In that page, I can see several options
("*" is a push button, "-" is a comment)
1. Installed Feature
* INSTALL FEATURE
2. System Update
- Please upload a valid update file
* CHOOSE FILE      * INSTALL
3. Log download
- Download all system logs as an encrypted artifact. This is typically requested for pport enquiries.
* DOWENLOAD LOGS
4. Factory Reset
- Warning: This will reset the system. Your.... Support first!
* FACTORY RESET
5. Move Robot to Transport Pose
-Warning: Please make sure to remove the endeffector for this and watch the not carefully
* MOVE TO PAACK POSE
6. Legal information
* OPEN-SOURCE LICENSES * GPL OFFER

Then, I don't have a FCI feature, am I right?

sgabl, 2019-03-24:
Yes, FCI feature is missing - please contact support with your serial number.

Junhyuck-Woo, 2019-03-25:
Thank you, it'll be handled this problem based on your advice.

closed 2019-07-05 by sgabl

manavkulshrestha, 2025-09-07:
Hi, I'm running into this same issue. However, I have FCI installed and FCI is enabled when I get the same "Connection timeout" message from libfranka