frankarobotics/libfranka #43: libfranka on Windows 10

MarcelBa, 2019-06-09:
Hello,

I try to run the "echo-robot-state"-example on a windows 10 PC. I downloaded libfranka-library, libfranka-common, the latest eigen-library and poco1.9.0. I'm able to compile the example in visual studio without getting an error message. But when I try to run the example I get the "UDP receive: Timeout"-message. 
- ping-check is ok
- robot controller is directly connected to the pc
- FCI is installed
Same example works fine on the same PC running Ubuntu. If you need more detailed informations, tell me.

Thank you.
MarcelBa

fwalch, 2019-06-12:
Does turning off the Windows firewall help?

MarcelBa, 2019-06-13:
Hello fwalch,

thanks for your answer. That was too easy 😆 now it works. Thank you!

fwalch, 2019-06-14:
You're welcome!

closed 2019-06-14 by fwalch