frankarobotics/libfranka #45: Use libfranka in simulation and without real robot

kargarisaac, 2019-06-17:
Hello,

Is it possible to use libfranka with simulation and without connecting to a real robot? I see in codes that it needs robot-hostname. 

What is your suggestion for controlling the panda in simulation? There are a lot of useful codes and controllers in libfranka.

Thank you

sgabl, 2019-07-05:
Hi, without the real robot you won't have access to the model library. Further information can be found there: [1](https://erdalpekel.de/?p=55), [2](https://github.com/frankaemika/franka_ros/issues/44) and [3](https://www.franka-community.de/t/urdf-model-of-the-panda-robot-including-masses-for-simulaton-in-gazebo/200/8)

closed 2019-07-05 by sgabl

Tadinu, 2025-08-14:
Just for reference, this is a recent drop-in simulation alternative in python (not the binding for not using this repo)
https://github.com/BarisYazici/libfranka-sim