frankarobotics/libfranka #47: Virtual Wall Example

djl11, 2019-07-04:
Can a minimal example be provided of how to use the Virtual Wall feature, as well as usage examples of the research_interface:: robot::GetCartesianLimit? Or are these features not functional at this time?

sgabl, 2019-07-05:
At the moment, there is no possibility to set the virtual walls.

djl11, 2019-07-05:
Okay, thanks for letting me know, I appreciate your fast response!
In my opinion, this would be a great and important addition to the library.

As far as I can tell, there is no simple way of enforcing cartesian position constrains, in the same manner in which velocity, acceleration, and force constraints are enforced, with errors being thrown which kill control to the robot for protection. Please correct me if I am wrong?

If there is a fixed known configuration of the robot with respect to a table, for example, and a custom closed-loop controller is designed, with no use of MoveIt for planning and collision avoidance, then it would be great if there was the option for a simple safety constraint which could prevent end-effector collisions with the table.

I understand there is no estimate on the timeline for this functionality being added.
I just wanted to make a case for the usefulness of this feature from my perspective! :)

Thanks

megasus, 2020-05-01:
Any update on this? Addition of a virtual wall feature would be very appreciated.

AndreasKuhner, 2024-09-26:
Hi @megasus ,
we don't necessarily have a 'virtual wall' feature but with our 5.6.0 release, you have the chance to set a SLP-C box to stop the robot entering a volume in Cartesian space. 

However, this is a safety feature which can be configured via Desk and there is no direct way of interacting with that via FCI.

closed 2024-09-26 by AndreasKuhner