frankarobotics/libfranka #48: Camera-on-hand not accounted for when running examples

aPonza, 2019-07-11:
I've mounted a hand camera between flange and hand and I can't run examples without eventually running into a reflex state due to (I think) the controller not accounting the extra weight. I can e.g. run
```
$ cd libfranka/build
$ ./examples/communication_test <robot_ip>
```
and I see the arm moving to the home position, then performing the test while the "forearm" falls towards the base. The test completes but I end up in self collision (without of course the robot actually physically touching itself).

fwalch, 2019-07-11:
If you have a fixed additional weight mounted to the flange/end-effector, you should configure this in Desk in Settings -> End-Effector.

For a dynamic payload, see [`franka::Robot::setLoad`](https://frankaemika.github.io/libfranka/classfranka_1_1Robot.html#afcb708df10f24563dbcf7d5b907b4a15).

aPonza, 2019-07-11:
Cheers, thank you.

closed 2019-07-11 by aPonza