frankarobotics/libfranka #49: Is there a way to do continuous path movements with libfranka?

MisterChi8, 2019-08-02:


fwalch, 2019-08-05:
I'm not sure what you are trying to do exactly, but do the [documentation](https://frankaemika.github.io/docs/libfranka.html#realtime-commands) and the provided [examples](https://github.com/frankaemika/libfranka/tree/master/examples) answer your question?

MisterChi8, 2019-08-05:
Hi, thanks for the reply. Unfortunately, the documentation and the examples do not answer my question.

Explanation for what I mean with "continuous path movements" (in german): "Bahnsteuerungen (continuous path) werden dort eingesetzt, wo der Bahnverlauf bei Montage- oder Bearbeitungsprozessen für die Lösung der Problemstellung erforderlich sind. Dieser Bahnverlauf bezieht sich üblicherweise auf den Werkzeugmittelpunkt (TCP). Ein Aufgabe kann z. B. sein, eine dosierfähige Dichtmasse aufzutragen. Die Bahngeschwindigkeit des Werkzeugmittelpunktes, das ist in dem Fall die Dosiernadel, muß mit einer konstanten Bahngeschwindigkeit bewegt werden, damit die Dichtmasse gleichmäßig aufgetragen wird.
Weil die Achsen in definierter örtlicher und zeitlicher Abhängigkeit während des Bahnverlaufs stehen, stellt die Bahnsteuerung höhere Anforderungen an die Rechnerleistung als bei einer PTP-Steuerung. Selbstverständlich beherrscht eine Bahnsteuerung die einfache Bewegungsart PTP.
Jede Bahn oder Teilbahn muß in kartesische Bahnstützpunkte zerlegt werden. Ergänzt werden die Bewegungsbahnen durch die Bahnform (Linear-, Kreis-, Polynombahnen bzw. interpolierte Bahnen), kartesische Geschwindigkeit, Beschleunigung, usw.
Eine Linearbahn ist eine Gerade (in einer Ebene) zwischen Anfangs- und Endpunkt. Bei einer Kreisbahn steht der Normalvektor senkrecht auf der Ebene, die von dem Kreis aufgespannt wird. Ein Kreisbogen wird durch drei Stützpunkte beschrieben. Der Anfangspunkt darf nicht auf den Endpunkt fallen, da dann der Bogen nicht beschreibbar ist!" (http://forum-sondermaschinenbau.de/seiten/Bewegungsarten_Robotersteuerung.htm)

Can I do continuous path movements (Bahnsteuerung) with libfranka? Or do I need ROS?

AndreasKuhner, 2024-09-26:
Hi @MisterChi8 ,
if the question is still relevant.. then yes, ROS might be the best choice to get something more quickly.

Cheers,
Andreas

MisterChi8, 2024-09-26:
Not relevant anymore, but still thank you 😄