frankarobotics/libfranka #50: Package Lost on Windows

MarcelBa, 2019-08-22:
Hello.
I'm using libfranka 0.7.1 on a Windows 10 PC. When i run the example "generate_cartesian_pose_motion" sometimes it works fine (20%) and sometimes not, caused by packages lost. The communication_test did not even start caused by packages lost. There is a second hard drive on the same PC running Ubuntu. When i run the examples on Ubuntu everything is fine (same hardware).

Used hardware:
- Intel Core i5-4690 3.5 GHz
- Intel PCI Ethernet Adapter

I wrote my own small application in Visual Studio 2017 to move the robot and I still get the same problems even if i run the robot.control-command in a native c++ thread.
Windows firewall is disabled and there is no other network connection. I tried to change the priority of the running task (in taskmanager) to "realtime" but it does not solve the problem. The PC is directly connected to the controller.
Are there any properties I have to set in Windows or Visual Studio to use libfranka with realtime communication? Or is it possible to use libfranka in kernel mode?

Thanks!
MarcelBa

poerob, 2020-09-29:
Hi,
i'm afraid this is not a satisfying answer to your problem - but i also tried to get libfranka on windows to work but had no success.  Non-Realtime Commands worked quite good, while realtime commands always failed. I tried different hard- and software combinations and also setting the priority but i always had timeouts, package loss and so on. Finally i switched to linux and now everything works fine - so thats my recommendation at this point.

AndreasKuhner, 2024-09-26:
Hi ghost,
as @poerob mentions, linux might be just easier at this point.. :) But if you have any further input about the usage for Windows, we would be happy to hear about it.

Cheers,
Andreas

closed 2024-09-26 by AndreasKuhner