frankarobotics/libfranka #53: Calibrated DH Parameters

John-HarringtonNZ, 2019-10-07:
The Panda-arm I'm using has a significant Cartesian error intrinsic to itself, where the expected point of the end effector is deviating >1cm from the expected position based on the standard DH parameters. 

I'm assuming the robots are all calibrated before shipping, and am hoping that the associated controller hardware has the calibrated DH parameters that would account for this manufacturing tolerance. Currently the jacobian created from the standard DH parameters and the jacobian from the libfranka controller are identical, meaning that I believe that the DH parameters used in the robot controller are not the calibrated ones. Is there a way to access and use the calibrated DH parameters? Otherwise it seems impossible to use the robots to their specified accuracy.

Thompson104, 2020-01-26:
Any update on this issue?

AndreasKuhner, 2024-09-26:
Hi @John-HarringtonNZ ,
thank you for your feedback. Without disclosing to much internal information, while our robot has a very good repeatability in joint space, we are not that precise for the Cartesian space (as you figured out). Of course, we always try to improve but there is still room for improvement :)

Cheers,
Andreas

closed 2024-09-26 by AndreasKuhner