frankarobotics/libfranka #55: Compiling Libfranka on Windows 10

poerob, 2019-11-29:
Hello,

I'd like to test the experimental Support for Windows 10. I followd the instructions from your [Documentation ](https://frankaemika.github.io/docs/installation_windows.html) using the **Developer Command Prompt for VS**-Method, but i do not have success.

Using the latest lib-Version from git, the build process output looks like this:
```
C:\repos\libfranka\build>cmake -DPoco_DIR=C:\repos\vcpkg\installed\x86-windows\share\poco -DEigen3_DIR=C:\repos\vcpkg\installed\x86-windows\share\eigen3 -G Ninja ..
-- The CXX compiler identification is MSVC 19.23.28107.0
-- Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/BuildTools/VC/Tools/MSVC/14.23.28105/bin/Hostx86/x86/cl.exe
-- Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/BuildTools/VC/Tools/MSVC/14.23.28105/bin/Hostx86/x86/cl.exe -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found Eigen3: C:/repos/vcpkg/installed/x86-windows/include/eigen3
-- Looking for C++ include pthread.h
-- Looking for C++ include pthread.h - not found
-- Found Threads: TRUE
-- The CXX compiler identification is MSVC 19.23.28107.0
-- Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/BuildTools/VC/Tools/MSVC/14.23.28105/bin/Hostx86/x86/cl.exe
-- Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/BuildTools/VC/Tools/MSVC/14.23.28105/bin/Hostx86/x86/cl.exe -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Configuring done
-- Generating done
-- Build files have been written to: C:/repos/libfranka/build/googletest-download
[1/9] Creating directories for 'googletest'
[2/9] Performing download step (download, verify and extract) for 'googletest'
-- verifying file...
       file='C:/repos/libfranka/3rdparty/googletest-1.8.0.tar.gz'
-- File already exists and hash match (skip download):
  file='C:/repos/libfranka/3rdparty/googletest-1.8.0.tar.gz'
  SHA256='58a6f4277ca2bc8565222b3bbd58a177609e9c488e8a72649359ba51450db7d8'
-- extracting...
     src='C:/repos/libfranka/3rdparty/googletest-1.8.0.tar.gz'
     dst='C:/repos/libfranka/3rdparty/googletest'
-- extracting... [tar xfz]
-- extracting... [analysis]
-- extracting... [rename]
-- extracting... [clean up]
-- extracting... done
[3/9] No update step for 'googletest'
[4/9] No patch step for 'googletest'
[5/9] No configure step for 'googletest'
[6/9] No build step for 'googletest'
[7/9] No install step for 'googletest'
[8/9] No test step for 'googletest'
[9/9] Completed 'googletest'
-- The C compiler identification is MSVC 19.23.28107.0
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/BuildTools/VC/Tools/MSVC/14.23.28105/bin/Hostx86/x86/cl.exe
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/BuildTools/VC/Tools/MSVC/14.23.28105/bin/Hostx86/x86/cl.exe -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Found PythonInterp: C:/Anaconda3/python.exe (found version "3.6.5")
-- Performing Test HAVE_ASan
-- Performing Test HAVE_ASan - Failed
-- Performing Test HAVE_UBSan
-- Performing Test HAVE_UBSan - Failed
-- Performing Test HAVE_TSan
-- Performing Test HAVE_TSan - Failed
-- Configuring done
-- Generating done
-- **Build files have been written to: C:/repos/libfranka/build**
```
Also the CMakeError.log shows some Errors. I attached both CMakeError.log and CMakeOutput.log for you:
[CMakeOutput.log](https://github.com/frankaemika/libfranka/files/3904061/CMakeOutput.log)
[... 124 lines cut ...]
[6/9] No build step for 'googletest'
[7/9] No install step for 'googletest'
[8/9] No test step for 'googletest'
[9/9] Completed 'googletest'
-- The C compiler identification is MSVC 19.23.28107.0
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/BuildTools/VC/Tools/MSVC/14.23.28105/bin/Hostx86/x86/cl.exe
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/BuildTools/VC/Tools/MSVC/14.23.28105/bin/Hostx86/x86/cl.exe -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Found PythonInterp: C:/Anaconda3/python.exe (found version "3.6.5")
-- Performing Test HAVE_ASan
-- Performing Test HAVE_ASan - Failed
-- Performing Test HAVE_UBSan
-- Performing Test HAVE_UBSan - Failed
-- Performing Test HAVE_TSan
-- Performing Test HAVE_TSan - Failed
-- Configuring incomplete, errors occurred!
See also "C:/repos/libfranka-0.6.0/build/CMakeFiles/CMakeOutput.log".
See also "C:/repos/libfranka-0.6.0/build/CMakeFiles/CMakeError.log"`

Do you have any suggestions? 

Thank you in advance!

fwalch, 2019-11-29:
I can't see any error in the first build log you posted. You need to run `ninja` or `cmake --build .` after CMake generated the build files.

As for this error:

```
CMake Error at CMakeLists.txt:48 (add_subdirectory):
  The source directory

    C:/repos/libfranka-0.6.0/common

  does not contain a CMakeLists.txt file.
```

Please run `git submodule update --init` or use `git clone --recursive` to get the submodule.

poerob, 2019-11-29:
Thank you for your fast reply.

Did you check the attached CMakeError.log ? There are some more Errors, whoch seem to come from ninja:

`Determining if the include file pthread.h exists failed with the following output:
Change Dir: C:/repos/libfranka/build/CMakeFiles/CMakeTmp

Run Build Command(s):C:/PROGRA~2/MIB055~1/2019/BUILDT~1/Common7/IDE/COMMON~1/MICROS~1/CMake/Ninja/ninja.exe cmTC_be4c4 
[1/2] Building CXX object CMakeFiles\cmTC_be4c4.dir\CheckIncludeFile.cxx.obj

FAILED: CMakeFiles/cmTC_be4c4.dir/CheckIncludeFile.cxx.obj 

C:\PROGRA~2\MIB055~1\2019\BUILDT~1\VC\Tools\MSVC\1423~1.281\bin\Hostx86\x86\cl.exe  /nologo /TP   /DWIN32 /D_WINDOWS /W3 /GR /EHsc  /MDd /Zi /Ob0 /Od /RTC1 /showIncludes /FoCMakeFiles\cmTC_be4c4.dir\CheckIncludeFile.cxx.obj /FdCMakeFiles\cmTC_be4c4.dir\ /FS -c CheckIncludeFile.cxx

CheckIncludeFile.cxx(1): fatal error C1083: Datei (Include) kann nicht geöffnet werden: "pthread.h": No such file or directory
**ninja: build stopped: subcommand failed.**

Performing C++ SOURCE FILE Test HAVE_ASan failed with the following output:
Change Dir: C:/repos/libfranka/build/CMakeFiles/CMakeTmp

Run Build Command(s):C:/PROGRA~2/MIB055~1/2019/BUILDT~1/Common7/IDE/COMMON~1/MICROS~1/CMake/Ninja/ninja.exe cmTC_e9125 
[1/2] Building CXX object CMakeFiles\cmTC_e9125.dir\src.cxx.obj

FAILED: CMakeFiles/cmTC_e9125.dir/src.cxx.obj 

C:\PROGRA~2\MIB055~1\2019\BUILDT~1\VC\Tools\MSVC\1423~1.281\bin\Hostx86\x86\cl.exe  /nologo /TP   /DWIN32 /D_WINDOWS /W3 /GR /EHsc -DHAVE_ASan /MDd /Zi /Ob0 /Od /RTC1   -g -O1 -fsanitize=address /showIncludes /FoCMakeFiles\cmTC_e9125.dir\src.cxx.obj /FdCMakeFiles\cmTC_e9125.dir\ /FS -c src.cxx

cl : Befehlszeile warning D9025 : "/Od" wird durch "/O1" überschrieben
cl : Befehlszeile error D8016 : Die Befehlszeilenoptionen /RTC1 und /O1 sind inkompatibel.
**ninja: build stopped: subcommand failed.**

Source file was:
int main() { return 0; }
Performing C++ SOURCE FILE Test HAVE_UBSan failed with the following output:
Change Dir: C:/repos/libfranka/build/CMakeFiles/CMakeTmp

Run Build Command(s):C:/PROGRA~2/MIB055~1/2019/BUILDT~1/Common7/IDE/COMMON~1/MICROS~1/CMake/Ninja/ninja.exe cmTC_93388 
[1/2] Building CXX object CMakeFiles\cmTC_93388.dir\src.cxx.obj

FAILED: CMakeFiles/cmTC_93388.dir/src.cxx.obj 

C:\PROGRA~2\MIB055~1\2019\BUILDT~1\VC\Tools\MSVC\1423~1.281\bin\Hostx86\x86\cl.exe  /nologo /TP   /DWIN32 /D_WINDOWS /W3 /GR /EHsc -DHAVE_UBSan /MDd /Zi /Ob0 /Od /RTC1   -g -O1 -fsanitize=undefined -fno-sanitize=alignment -fno-sanitize-recover=undefined /showIncludes /FoCMakeFiles\cmTC_93388.dir\src.cxx.obj /FdCMakeFiles\cmTC_93388.dir\ /FS -c src.cxx

cl : Befehlszeile warning D9025 : "/Od" wird durch "/O1" überschrieben
cl : Befehlszeile error D8016 : Die Befehlszeilenoptionen /RTC1 und /O1 sind inkompatibel.
**ninja: build stopped: subcommand failed.**

Source file was:
int main() { return 0; }
Performing C++ SOURCE FILE Test HAVE_TSan failed with the following output:
Change Dir: C:/repos/libfranka/build/CMakeFiles/CMakeTmp

Run Build Command(s):C:/PROGRA~2/MIB055~1/2019/BUILDT~1/Common7/IDE/COMMON~1/MICROS~1/CMake/Ninja/ninja.exe cmTC_cd6f9 
[1/2] Building CXX object CMakeFiles\cmTC_cd6f9.dir\src.cxx.obj

FAILED: CMakeFiles/cmTC_cd6f9.dir/src.cxx.obj 

C:\PROGRA~2\MIB055~1\2019\BUILDT~1\VC\Tools\MSVC\1423~1.281\bin\Hostx86\x86\cl.exe  /nologo /TP   /DWIN32 /D_WINDOWS /W3 /GR /EHsc -DHAVE_TSan /MDd /Zi /Ob0 /Od /RTC1   -g -O1 -fsanitize=thread /showIncludes /FoCMakeFiles\cmTC_cd6f9.dir\src.cxx.obj /FdCMakeFiles\cmTC_cd6f9.dir\ /FS -c src.cxx

cl : Befehlszeile warning D9025 : "/Od" wird durch "/O1" überschrieben
cl : Befehlszeile error D8016 : Die Befehlszeilenoptionen /RTC1 und /O1 sind inkompatibel.
**ninja: build stopped: subcommand failed.**

Source file was:
int main() { return 0; }
`
As i said before, there are no binaries. I can do a wildcard search for *.exe files in the whole libfranka-folder after running cmake and ninja not finding anything (or maybe some cmake stuff).

poerob, 2019-11-29:
Running cmake --build  . from build folder instead of ninja returns the following error:

`[1/83] Building CXX object CMakeFiles\franka.dir\src\duration.cpp.obj
cl : Befehlszeile warning D9025 : "/W3" wird durch "/W0" überschrieben
[2/83] Building CXX object CMakeFiles\franka.dir\src\control_tools.cpp.obj
FAILED: CMakeFiles/franka.dir/src/control_tools.cpp.obj
C:\PROGRA~2\MIB055~1\2019\BUILDT~1\VC\Tools\MSVC\1423~1.281\bin\Hostx86\x86\cl.exe  /nologo /TP -DNOMINMAX -DPOCO_NO_AUTOMATIC_LIBS -D_USE_MATH_DEFINES -Dfranka_EXPORTS -I..\include -I..\common\include -IC:\repos\vcpkg\installed\x86-windows\include -IC:\repos\vcpkg\installed\x86-windows\include\eigen3 /DWIN32 /D_WINDOWS /W3 /GR /EHsc /MDd /Zi /Ob0 /Od /RTC1   /W0 -std:c++14 /showIncludes /FoCMakeFiles\franka.dir\src\control_tools.cpp.obj /FdCMakeFiles\franka.dir\ /FS -c ..\src\control_tools.cpp
cl : Befehlszeile warning D9025 : "/W3" wird durch "/W0" überschrieben
..\src\control_tools.cpp(32): error C2065: "DWORD": nichtdeklarierter Bezeichner
..\src\control_tools.cpp(32): error C2146: Syntaxfehler: Fehlendes ";" vor Bezeichner "error_id"
..\src\control_tools.cpp(32): error C2065: "error_id": nichtdeklarierter Bezeichner
..\src\control_tools.cpp(32): error C3861: "GetLastError": Bezeichner wurde nicht gefunden.
..\src\control_tools.cpp(33): error C2065: "LPSTR": nichtdeklarierter Bezeichner
..\src\control_tools.cpp(33): error C2146: Syntaxfehler: Fehlendes ";" vor Bezeichner "buffer"
..\src\control_tools.cpp(33): error C2065: "buffer": nichtdeklarierter Bezeichner
..\src\control_tools.cpp(35): error C2065: "FORMAT_MESSAGE_ALLOCATE_BUFFER": nichtdeklarierter Bezeichner
..\src\control_tools.cpp(35): error C2065: "FORMAT_MESSAGE_FROM_SYSTEM": nichtdeklarierter Bezeichner
..\src\control_tools.cpp(35): error C2065: "FORMAT_MESSAGE_IGNORE_INSERTS": nichtdeklarierter Bezeichner
..\src\control_tools.cpp(36): error C2065: "error_id": nichtdeklarierter Bezeichner
..\src\control_tools.cpp(36): error C2065: "LANG_NEUTRAL": nichtdeklarierter Bezeichner
..\src\control_tools.cpp(36): error C2065: "SUBLANG_DEFAULT": nichtdeklarierter Bezeichner
..\src\control_tools.cpp(36): error C3861: "MAKELANGID": Bezeichner wurde nicht gefunden.
..\src\control_tools.cpp(36): error C2065: "LPSTR": nichtdeklarierter Bezeichner
..\src\control_tools.cpp(36): error C2065: "buffer": nichtdeklarierter Bezeichner
..\src\control_tools.cpp(34): error C3861: "FormatMessageA": Bezeichner wurde nicht gefunden.
..\src\control_tools.cpp(37): error C2065: "buffer": nichtdeklarierter Bezeichner
..\src\control_tools.cpp(40): error C3861: "GetCurrentProcess": Bezeichner wurde nicht gefunden.
..\src\control_tools.cpp(40): error C2065: "REALTIME_PRIORITY_CLASS": nichtdeklarierter Bezeichner
..\src\control_tools.cpp(40): error C3861: "SetPriorityClass": Bezeichner wurde nicht gefunden.
..\src\control_tools.cpp(48): error C3861: "GetCurrentThread": Bezeichner wurde nicht gefunden.
..\src\control_tools.cpp(48): error C2065: "THREAD_PRIORITY_TIME_CRITICAL": nichtdeklarierter Bezeichner
..\src\control_tools.cpp(48): error C3861: "SetThreadPriority": Bezeichner wurde nicht gefunden.
[3/83] Building CXX object CMakeFiles\franka.dir\src\errors.cpp.obj
cl : Befehlszeile warning D9025 : "/W3" wird durch "/W0" überschrieben
[4/83] Building CXX object CMakeFiles\franka.dir\src\control_types.cpp.obj
cl : Befehlszeile warning D9025 : "/W3" wird durch "/W0" überschrieben
[5/83] Building CXX object CMakeFiles\franka.dir\src\gripper_state.cpp.obj
cl : Befehlszeile warning D9025 : "/W3" wird durch "/W0" überschrieben
[6/83] Building CXX object CMakeFiles\franka.dir\src\exception.cpp.obj
cl : Befehlszeile warning D9025 : "/W3" wird durch "/W0" überschrieben
[7/83] Building CXX object CMakeFiles\franka.dir\src\control_loop.cpp.obj
cl : Befehlszeile warning D9025 : "/W3" wird durch "/W0" überschrieben
[8/83] Building CXX object CMakeFiles\franka.dir\src\library_loader.cpp.obj
cl : Befehlszeile warning D9025 : "/W3" wird durch "/W0" überschrieben
[9/83] Building CXX object CMakeFiles\franka.dir\src\library_downloader.cpp.obj
cl : Befehlszeile warning D9025 : "/W3" wird durch "/W0" überschrieben
ninja: build stopped: subcommand failed.`

fwalch, 2019-11-29:
The entries from the CMake error log should be just the results of the feature tests that CMake runs (e.g. `Looking for C++ include pthread.h - not found`), but these aren't required for the build to succeed.

In your second comment, it seems that we accidentally removed an include that is required on Windows. Can you apply the following changes and try compiling again:

```diff
diff --git a/src/control_tools.cpp b/src/control_tools.cpp
index c95a7d6..44ac5de 100644
--- a/src/control_tools.cpp
+++ b/src/control_tools.cpp
@@ -8,6 +8,12 @@
 
 #include "platform.h"
 
+#ifdef LIBFRANKA_WINDOWS
+#include <Windows.h>
+#else
+#include <pthread.h>
+#endif
+
 // `using std::string_literals::operator""s` produces a GCC warning that cannot be disabled, so we
 // have to use `using namespace ...`.
 // See https://gcc.gnu.org/bugzilla/show_bug.cgi?id=65923#c0
```

poerob, 2019-11-29:
I tried both adding 

`
#ifdef LIBFRANKA_WINDOWS
#include <Windows.h>
#else
#include <pthread.h>
#endif
`
and just

`
#include <Windows.h>
`
to src/control_tools.cpp, but got following errors in the command line:

ninja:
[1/83] Building CXX object CMakeFiles\franka.dir\src\duration.cpp.obj
cl : Befehlszeile warning D9025 : "/W3" wird durch "/W0" überschrieben
[2/83] Building CXX object CMakeFiles\franka.dir\src\control_types.cpp.obj
cl : Befehlszeile warning D9025 : "/W3" wird durch "/W0" überschrieben
[3/83] Building CXX object CMakeFiles\franka.dir\src\errors.cpp.obj
cl : Befehlszeile warning D9025 : "/W3" wird durch "/W0" überschrieben
[4/83] Building CXX object CMakeFiles\franka.dir\src\exception.cpp.obj
cl : Befehlszeile warning D9025 : "/W3" wird durch "/W0" überschrieben
[5/83] Building CXX object CMakeFiles\franka.dir\src\control_loop.cpp.obj
cl : Befehlszeile warning D9025 : "/W3" wird durch "/W0" überschrieben
[6/83] Building CXX object CMakeFiles\franka.dir\src\control_tools.cpp.obj
FAILED: CMakeFiles/franka.dir/src/control_tools.cpp.obj
C:\PROGRA~2\MIB055~1\2019\BUILDT~1\VC\Tools\MSVC\1423~1.281\bin\Hostx86\x86\cl.exe  /nologo /TP -DNOMINMAX -DPOCO_NO_AUTOMATIC_LIBS -D_USE_MATH_DEFINES -Dfranka_EXPORTS -I..\include -I..\common\include -IC:\repos\vcpkg\installed\x86-windows\include -IC:\repos\vcpkg\installed\x86-windows\include\eigen3 /DWIN32 /D_WINDOWS /W3 /GR /EHsc /MDd /Zi /Ob0 /Od /RTC1   /W0 -std:c++14 /showIncludes /FoCMakeFiles\franka.dir\src\control_tools.cpp.obj /FdCMakeFiles\franka.dir\ /FS -c ..\src\control_tools.cpp
cl : Befehlszeile warning D9025 : "/W3" wird durch "/W0" überschrieben
..\src\control_tools.cpp(9): error C2001: Zeilenvorschub in Konstante.
[7/83] Building CXX object CMakeFiles\franka.dir\src\gripper_state.cpp.obj
cl : Befehlszeile warning D9025 : "/W3" wird durch "/W0" überschrieben
[8/83] Building CXX object CMakeFiles\franka.dir\src\logger.cpp.obj
cl : Befehlszeile warning D9025 : "/W3" wird durch "/W0" überschrieben
[9/83] Building CXX object CMakeFiles\franka.dir\src\gripper.cpp.obj
cl : Befehlszeile warning D9025 : "/W3" wird durch "/W0" überschrieben
[10/83] Building CXX object CMakeFiles\franka.dir\src\library_loader.cpp.obj
cl : Befehlszeile warning D9025 : "/W3" wird durch "/W0" überschrieben
[11/83] Building CXX object CMakeFiles\franka.dir\src\library_downloader.cpp.obj
cl : Befehlszeile warning D9025 : "/W3" wird durch "/W0" überschrieben
[12/83] Building CXX object CMakeFiles\franka.dir\src\lowpass_filter.cpp.obj
cl : Befehlszeile warning D9025 : "/W3" wird durch "/W0" überschrieben
[13/83] Building CXX object CMakeFiles\franka.dir\src\load_calculations.cpp.obj
cl : Befehlszeile warning D9025 : "/W3" wird durch "/W0" überschrieben
ninja: build stopped: subcommand failed.

i also cleaned up the build folder, ran cmake and finally cmake --build . 
This is the result:
`[1/83] Building CXX object CMakeFiles\franka.dir\src\duration.cpp.obj
cl : Befehlszeile warning D9025 : "/W3" wird durch "/W0" überschrieben
[2/83] Building CXX object CMakeFiles\franka.dir\src\control_types.cpp.obj
cl : Befehlszeile warning D9025 : "/W3" wird durch "/W0" überschrieben
[3/83] Building CXX object CMakeFiles\franka.dir\src\errors.cpp.obj
cl : Befehlszeile warning D9025 : "/W3" wird durch "/W0" überschrieben
[4/83] Building CXX object CMakeFiles\franka.dir\src\exception.cpp.obj
cl : Befehlszeile warning D9025 : "/W3" wird durch "/W0" überschrieben
[5/83] Building CXX object CMakeFiles\franka.dir\src\control_loop.cpp.obj
cl : Befehlszeile warning D9025 : "/W3" wird durch "/W0" überschrieben
[6/83] Building CXX object CMakeFiles\franka.dir\src\gripper_state.cpp.obj
cl : Befehlszeile warning D9025 : "/W3" wird durch "/W0" überschrieben
[7/83] Building CXX object CMakeFiles\franka.dir\src\control_tools.cpp.obj
FAILED: CMakeFiles/franka.dir/src/control_tools.cpp.obj
C:\PROGRA~2\MIB055~1\2019\BUILDT~1\VC\Tools\MSVC\1423~1.281\bin\Hostx86\x86\cl.exe  /nologo /TP -DNOMINMAX -DPOCO_NO_AUTOMATIC_LIBS -D_USE_MATH_DEFINES -Dfranka_EXPORTS -I..\include -I..\common\include -IC:\repos\vcpkg\installed\x86-windows\include -IC:\repos\vcpkg\installed\x86-windows\include\eigen3 /DWIN32 /D_WINDOWS /W3 /GR /EHsc /MDd /Zi /Ob0 /Od /RTC1   /W0 -std:c++14 /showIncludes /FoCMakeFiles\franka.dir\src\control_tools.cpp.obj /FdCMakeFiles\franka.dir\ /FS -c ..\src\control_tools.cpp
cl : Befehlszeile warning D9025 : "/W3" wird durch "/W0" überschrieben
..\src\control_tools.cpp(9): error C2001: Zeilenvorschub in Konstante.
[8/83] Building CXX object CMakeFiles\franka.dir\src\library_loader.cpp.obj
cl : Befehlszeile warning D9025 : "/W3" wird durch "/W0" überschrieben
[9/83] Building CXX object CMakeFiles\franka.dir\src\gripper.cpp.obj
cl : Befehlszeile warning D9025 : "/W3" wird durch "/W0" überschrieben
[10/83] Building CXX object CMakeFiles\franka.dir\src\log.cpp.obj
cl : Befehlszeile warning D9025 : "/W3" wird durch "/W0" überschrieben
[11/83] Building CXX object CMakeFiles\franka.dir\src\logger.cpp.obj
cl : Befehlszeile warning D9025 : "/W3" wird durch "/W0" überschrieben
[12/83] Building CXX object CMakeFiles\franka.dir\src\library_downloader.cpp.obj
cl : Befehlszeile warning D9025 : "/W3" wird durch "/W0" überschrieben
[13/83] Building CXX object CMakeFiles\franka.dir\src\network.cpp.obj
cl : Befehlszeile warning D9025 : "/W3" wird durch "/W0" überschrieben
[14/83] Building CXX object CMakeFiles\franka.dir\src\load_calculations.cpp.obj
cl : Befehlszeile warning D9025 : "/W3" wird durch "/W0" überschrieben
ninja: build stopped: subcommand failed.`

poerob, 2019-11-29:
Edit: My fault, accidentally deleted a character in the include section. Now compiling works fine and i executable binaries. I'll test them next week.

Thank you for your help, have a nice weekend!

closed 2019-12-01 by poerob