frankarobotics/libfranka #56: Example Joint Impendance Control fails while downloading model library

poerob, 2020-01-10:
Hello,

running "joint_impendance_control" from the examples folder always fails with 
"libfranka: Server reports error when loading model library".
Other functions like generate_elbow_motion or generate_cartesian_pose_motion run flawlessly.

My Environment is: A Raspberry Pi 4 running Raspbian with RT-Patch and libfranka library build from source. The Firewall is turned off, the pi is connected to the panda-robot by the on-board ethernet plug and an 0.5m ethernet cable. Do you have an idea how to fix the problem?

Thank you in advance!

LorisR, 2020-01-20:
I'm having the same issue.
In my case, when in the program the line "franka::Model model = robot.loadModel()" is called I do not have any error but the robot freezes, the communication is no longer available even from the web interface and I have to switch the controller off pushing the button.

fwalch, 2020-01-20:
The model library for ARM is only included from system version 3.1.0+. This is mentioned in the libfranka changelog entry for 0.7.0, but we will think about how to communicate this better.

LorisR, 2020-01-20:
I'm actually using system version 3.1.0 but the issue is there.

AndreasKuhner, 2024-09-26:
Hey @LorisR ,
do you still have this issue with newer versions of the system? If yes, please open up a new issue. 

Cheers,
Andreas

(being here to clean up a bit...)

closed 2024-09-26 by AndreasKuhner