frankarobotics/libfranka #60: [communication_constraints_violation] error.

coopawhite, 2020-02-12:
Hi,

Im currently having issues running any of the examples provided with the libfranka library. I have received this error on both the source install and ros install methods. 

I am able to control the robot via the desk using static ip. 
 
I am also able to complete the echo_robot_state and print_joint_poses examples. 

Does anyone have any ideas where this error stems from?
```
 rosrun libfranka communication_test 172.16.0.2
WARNING: This example will move the robot! Please make sure to have the user stop button at hand!
Press Enter to continue...

libfranka: Move command aborted: motion aborted by reflex! ["communication_constraints_violation"]
control_command_success_rate: 0.78
```

gavanderhoorn, 2020-02-12:
Can you check you're still starting the PREEMPT kernel?

I've had this happen to a student, where after an update they booted the vanilla kernel and things stopped working.

coopawhite, 2020-02-12:
> Can you check you're still starting the PREEMPT kernel?
> 
> I've had this happen to a student, where after an update they booted the vanilla kernel and things stopped working.

Hi, 

Thanks for your reply. I have ensured that I boot the real time kernel from the grub loader. 

You mentioned your student, is there any chance you have some teaching material based on the Franka? I'm really struggling to come to grips with it and no one at my university has used one before.

zbw0520, 2020-02-12:
I met the same issue
```
C:\vcpkg-master\libfranka\build\x64-debug\examples>communication_test.exe 172.16.0.2
WARNING: This example will move the robot! Please make sure to have the user stop button at hand!
Press Enter to continue...

Finished moving to initial joint configuration.

Starting communication test.
libfranka: Move command aborted: motion aborted by reflex! ["communication_constraints_violation"]
control_command_success_rate: 0.7821 packets lost in a row in the last sample: 1
```
Hope there is some solution to it.

gavanderhoorn, 2020-02-12:
> You mentioned your student, is there any chance you have some teaching material based on the Franka? I'm really struggling to come to grips with it and no one at my university has used one before.

No, I don't have anything like that I'm afraid.

b3nE-3010, 2020-11-17:
Hi,
i have the same Problem.

The PREEMPT kernel is runing

sgabl, 2020-11-17:
> Hi,
> i have the same Problem.
> 
> The PREEMPT kernel is runing

Hi, probably the steps described [here](https://frankaemika.github.io/docs/troubleshooting.html#motion-stopped-due-to-discontinuities-or-communication-constraints-violation) help.

siweiyong, 2021-03-24:
> Hi,
>
> Im currently having issues running any of the examples provided with the libfranka library. I have received this error on both the source install and ros install methods.
>
> I am able to control [...]

Hi,
Have you solved your problems? i met the same problem, and I have tested all the possible reasons regarding the Internet.

Savitakendre, 2021-04-08:
> Hi,
> Have you solved your problems? i met the same problem, and I have tested all the possible reasons regarding the Internet.

Hello,

I am also having the same issue and did all the steps mentioned in Troubleshooting of that error. Could anyone please help me with it.
I am using Libfranka 0.7.1 on realtime PREEMPT Kernel. 

Need urgent help!

sgabl, 2021-04-08:
Hi,

if you followed all the steps on the troubleshooting page, you should try a different network card / PC. From my experience, external network cards connected through USB lead to a too high delay. What are your results for the [simple ping test](https://frankaemika.github.io/docs/troubleshooting.html#simple-ping-tests).

Savitakendre, 2021-04-08:
Hello,

Thank you for your reply. The ping test gives me following result:

10000 packets transmitted, 10000 received, 0% packet loss, time 10057ms
rtt min/avg/max/mdev = 0.078/0.205/0.456/0.043 ms

which doesnt seem to be a problem. 

Also when I run communication test, there are always different errors depending on current position of the robot. Following are the errors which occures mostly:

1. joint_motion_generator_velocity_discontinuity
2. joint_motion_generator_acceleration_discontinuity
3. Motion aborted by reflex! [cartesian_reflex]

once the test worked too, but with 42 lost robot state. I couldnt understand the reason why it worked at that time only. i have not got any 100% success rate.

sgabl, 2021-04-08:
Hi,

the avg in the ping test looks ok! The max is high and could lead to a performance loss, but the communication test should go through.
When running the communication test, the robot should not move at all. This test just sends zero torques. Please bring the robot in a configuration like [visualized here](https://frankaemika.github.io/docs/_images/fci-architecture.png) and try the communication test again. When the robot moves, make sure, that you compensated the gripper correctly.

Savitakendre, 2021-04-08:
Hello Simon,

Thank you for a quick reply. Sure. I will try it tomorrow and see if that works.

Thanks for your help again!

Savitakendre, 2021-04-13:
Hello Simon,

I have tried out with your suggestion. For communication test, I executed the same code as provided and encountered the same error. What I did is just commented the initial configuration part as shown below:

```
try {
    franka::Robot robot(argv[1]);
    setDefaultBehavior(robot);
    // First move the robot to a suitable joint configuration
    /*std::array<double, 7> q_goal = {{0.215121,-0.789043,-0.353009,-2.3029,-0.321726,1.65329,-0.307532}};
    MotionGenerator motion_generator(0.5, q_goal);
    */
    std::cout << "WARNING: This example will move the robot! "
              << "Please make sure to have the user stop button at hand!" << std::endl
              << "Savita_checking if its at all changing or not" << std::endl
              << "Press Enter to continue..." << std::endl;
    std::cin.ignore();
    /*
    robot.control(motion_generator);
    */
    std::cout << "Finished moving to initial joint configuration." << std::endl << std::endl;
    std::cout << "Starting communication test." << std::endl;
```

And started the communication test. It got executed without error, But there are always some robot state losses between 20 to 80.

Following is the output of the communication test:
```
#8700 Current success rate: 0.34
#8800 Current success rate: 0.29
#8900 Current success rate: 0.28
#9000 Current success rate: 0.34
#9100 Current success rate: 0.26
#9200 Current success rate: 0.33
#9300 Current success rate: 0.32
#9400 Current success rate: 0.34
#9500 Current success rate: 0.37
#9600 Current success rate: 0.29
#9700 Current success rate: 0.36
#9800 Current success rate: 0.40
#9900 Current success rate: 0.28

Finished test, shutting down example

#######################################################
The control loop did not get executed 59 times in the
last 10000 milliseconds! (lost 59 robot states)

Control command success rate of 9941 samples: 
Max: 1.00
Avg: 0.34
Min: 0.20

WARNING: THIS SETUP IS PROBABLY NOT SUFFICIENT FOR FCI!
PLEASE TRY OUT A DIFFERENT PC / NIC
#######################################################

```
I am working on Linux lab3 5.4.19-rt11 #1 SMP PREEMPT_RT Fri Mar 12 10:08:11 CET 2021 x86_64 x86_64 x86_64 GNU/Linux. And I have build the libfranka 0.7.1.1 from source and I am not using ROS, instead directly running the code in Konsole with executables from build folder. Please let me know if there is something which I am missing here. I have tried many options but still the same result.

Savitakendre, 2021-04-13:
Any suggestions with respect to this will be really helpful. if someone have faced the similar problems to run the code.

sgabl, 2021-04-14:
Hi @Savitakendre,

reading the results you got, I think that only using a different network card, network cable or a complete different PC would make the FCI usable for you. I don't think that other software-wise optimizations are helpful.

Savitakendre, 2021-04-14:
Hello simon, 
Thank you for the suggestions. I am thinking to buy a new network card now and it would be really helpful if you could provide some suggestions on which one to buy, which has been already tested.

Savitakendre, 2021-04-15:
Hi Simon,
Just so you know, I have I219-LM ethernet network controller and also I210 and I218. Currently I have just tested it with I219-LM. Also there were some collegues of mine who suggested disabling the Hyper threading seemed to resolved their problem. But unfortunately it does not seem to work for us.

sgabl, 2021-04-15:
Hi @Savitakendre,

I have an I219-LM as well and get good results. Controlling the robot works with a different PC / Laptop?

Savitakendre, 2021-04-15:
Hello Simon,

I have not tried yet with other PC. I think it should work with I219 -LM
kernel. Probably there are some settings which needs to be done, or some
kind of optimisation, which I am not able to figure out yet.

Any further suggestions are always helpful.

longshike, 2021-11-03:
hi, have you solved this problem,  i miss the same problem with you

lukashermann, 2021-11-19:
Hello,
I also get the communication_constraints_violation error when I'm teleoperating the robot, it occurs approx. every 15 mins.
I am using the impedance motion generator of FrankX (https://github.com/pantor/frankx/blob/master/include/frankx/motion_impedance_generator.hpp), if this is of any relevance.
The workstation is connected directly to the FCI with a Realtek PCI Express Gigabit Ethernet Controller.
It runs Ubuntu 20.04 with 5.4.129-rt61.

The ping test shows no problem:
```
10000 packets transmitted, 10000 received, 0% packet loss, time 9998ms
rtt min/avg/max/mdev = 0.083/0.130/0.227/0.008 ms
```

The communication test however, says their are 69 lost robot states:
```
#######################################################
The control loop did not get executed 69 times in the
last 10000 milliseconds! (lost 69 robot states)

Control command success rate of 9931 samples: 
Max: 1.00
Avg: 0.95
Min: 0.65
#######################################################
```

Would really appreciate help with that!

marcbone, 2021-11-19:
69 missed robot states is not a problem. The minimum seems quite low though, resulting in a low average (should be around 0.97 on a reasonable computer). Assuming you set your governor to performance and compiled libfranka in release mode, I suggest you try out a different computer or try out an usb-to-ethernet converter and see if this solves the problem.

yijionglin, 2024-03-29:
For those who do more than just run the control loop, it's essential to check if there is anything that consumes CPU resources significantly. In my case, my program (utilizing deep neural networks and Franka control) ran well until I switched to a larger neural network, which consumes more CPU resources and consequently increases the round-trip time (the CPU frequency governer had been changed to 'performance').

AndreasKuhner, 2024-09-26:
Hi all,
we always try to improve. So, if there are further problems, please open a new issue.

Summary for common issues:
- Overloaded PC (e.g. those nasty _couts_)
- Network card not suitable
- Realtime kernel not active
- Overloaded network

Cheers,
Andreas

closed 2024-09-26 by AndreasKuhner