frankarobotics/libfranka #62: Realtime Kernel not building

rickstaa, 2020-04-25:
Just a quick heads-up about the real-time kernel instructions found in [your online documentation](https://frankaemika.github.io/docs/). These instructions do not seem to work anymore on my system (Ubuntu 18.04 with kernel versions higher than 5.0). I was, however, able to build the realtime kernel using the following steps:

1. Download the regular kernel and the patch from the [Linux Kernel archive](https://www.kernel.org/pub/linux/kernel/projects/rt/).
2. Verifying file integrity.
3. Unzip the Linux kernel `tar -xvf Linux-5.4.26.tar`.
4. Go into the kernel folder `cd Linux-5.4.28`.
5. Apply the patch `patch -p1 < ../patch-5.4.26-rt17.patch`.
6. Run `make menuconfig` and change the Preemption Model to 'Fully Preemptible Kernel (RT) (PREEMPT_RT_FULL) (NEW)'.
7. Disable lz4 compression `scripts/config --disable KERNEL_LZ4`.
8. Enable gzip compression `scripts/config --enable KERNEL_GZIP`.
9. Disable debug info `scripts/config --disable DEBUG_INFO`.
10. Run `make clean`.
11. Make the bzImage `make -j $(nproc) bzImage`.
12. Make the modules `make -j $(nproc) modules`.
13. Install the modules `make -j $(nproc) modules_install`.
14. Install the kernel `make -j $(nproc) install`.

museum-future, 2020-05-07:
Why did you disable compression?  What values did you use for zswap?

rickstaa, 2020-05-18:
@museum-future Thanks for looking into the problem. I disabled the LZ4 compression since when I kept, it enabled I gave me an error during compilation. Below I added a more detailed debug report of the problems I experienced when trying to compile the kernel using the steps provided by the [franka documentation](https://frankaemika.github.io/docs/).

#### Debug report
When trying to compile the rt kernel using the `fakeroot make -j4 deb-pkg` command that is given in the franka documentation I get the following error:

```
Makefile:613: include/config/auto.conf: No such file or directory
Makefile:659: include/config/auto.conf.cmd: No such file or directory
  HOSTCC  scripts/basic/fixdep
  HOSTCC  scripts/kconfig/conf.o
  HOSTCC  scripts/kconfig/confdata.o
  HOSTCC  scripts/kconfig/expr.o
  LEX     scripts/kconfig/lexer.lex.c
  YACC    scripts/kconfig/parser.tab.[ch]
  HOSTCC  scripts/kconfig/preprocess.o
  HOSTCC  scripts/kconfig/symbol.o
  HOSTCC  scripts/kconfig/lexer.lex.o
  HOSTCC  scripts/kconfig/parser.tab.o
  HOSTLD  scripts/kconfig/conf
scripts/kconfig/conf  --syncconfig Kconfig
  UPD     include/config/kernel.release
make clean
sh ./scripts/package/mkdebian
  TAR     linux-5.4.26-rt17.tar.gz
origversion=$(dpkg-parsechangelog -SVersion |sed 's/-[^-]*$//');\
        mv linux-5.4.26-rt17.tar.gz ../linux-5.4.26-rt17_${origversion}.orig.tar.gz
dpkg-buildpackage -r"fakeroot -u" -a$(cat debian/arch)  -i.git -us -uc
dpkg-buildpackage: warning: using a gain-root-command while being root
dpkg-buildpackage: info: source package linux-5.4.26-rt17
dpkg-buildpackage: info: source version 5.4.26-rt17-1
dpkg-buildpackage: info: source distribution bionic
dpkg-buildpackage: info: source changed by root <root@ricks-HP-ZBook-Studio-x360-G5-Ubutu18>
dpkg-buildpackage: info: host architecture amd64
 dpkg-source -i.git --before-build linux-5.4.26
 fakeroot -u debian/rules clean
fakeroot: FAKEROOTKEY set to 894722995
fakeroot: nested operation not yet supported
dpkg-buildpackage: error: fakeroot -u debian/rules clean subprocess returned exit status 1
scripts/Makefile.package:73: recipe for target 'deb-pkg' failed
make[1]: *** [deb-pkg] Error 1
Makefile:1425: recipe for target 'deb-pkg' failed
make: *** [deb-pkg] Error 2
[13:28:08] ricks@ricks-HP-ZBook-Studio-x360-G5-Ub
```

As I was not able to solve this problem I used the `make -j $(nproc) bzImage` command instead. While doing this when LZ4 compression is enabled I howver get the following error:

```
/bin/sh: 1: lz4c: not found
arch/x86/boot/compressed/Makefile:146: recipe for target 'arch/x86/boot/compressed/vmlinux.bin.lz4' failed
make[2]: *** [arch/x86/boot/compressed/vmlinux.bin.lz4] Error 127
make[2]: *** Deleting file 'arch/x86/boot/compressed/vmlinux.bin.lz4'
make[2]: *** Waiting for unfinished jobs....
arch/x86/boot/Makefile:112: recipe for target 'arch/x86/boot/compressed/vmlinux' failed
make[1]: *** [arch/x86/boot/compressed/vmlinux] Error 2
arch/x86/Makefile:284: recipe for target 'bzImage' failed
make: *** [bzImage] Error 2
```

An explanation about this issue can be found [here](https://askubuntu.com/questions/1194913/couldnt-compile-kernel-with-patch). At the time of writing, I did, however not know what caused the problem as I had both the `liblz4-1` and `liblz4-dev` packages installed. This problem, however, is solved by installing the `liblz4-tool` package. As a result, I can now compile the rt-kernel using the following (updated) steps:

1. Install the following necessary dependencies `apt-get install build-essential bc curl ca-certificates fakeroot gnupg2 libssl-dev lsb-release libelf-dev bison flex liblz4-tool` (The liblz4-tool dependency is added here since it is not installed by default on ubuntu 18.04). 
2. Download the regular kernel and the patch from the [Linux Kernel archive](https://www.kernel.org/pub/linux/kernel/projects/rt/).
3. Verifying file integrity.
4. Unzip the Linux kernel `tar -xvf Linux-5.4.26.tar`.
5. Go into the kernel folder `cd linux-5.4.26`.
6. Apply the patch `patch -p1 < ../patch-5.4.26-rt17.patch`.
7. Run `make menuconfig` and change the Preemption Model to 'Fully Preemptible Kernel (RT) (PREEMPT_RT_FULL) (NEW)'.
8. Disable debug info `scripts/config --disable DEBUG_INFO` (Not needed but recommended).
9. Run `make clean`.
10. Make the bzImage `make -j $(nproc) bzImage`.
11. Make the modules `make -j $(nproc) modules`.
12. Install the modules `make -j $(nproc) modules_install`.
13. Install the kernel `make -j $(nproc) install`.

#### Config file

The config file I use when trying to compile the realtime kernel can be found in this [gist](https://gist.github.com/rickstaa/bc3c8449892d591849a326efa5220ee5).

rickstaa, 2021-05-19:
@museum-future Just a heads up this issue still exists under ubuntu 20.04 with the 5.9 kernels. Another solution is found on https://stackoverflow.com/questions/51669724/install-rt-linux-patch-for-ubuntu.

Related to #101 so I think we can close one of the two issues.

rickstaa, 2021-05-19:
Looks like the Franka guide works if you replace the `fakeroot make -j4 deb-pkg` command with:

```bash
make -j$(nproc)
sudo make modules_install -j$(nproc)
sudo make install -j$(nproc)
```

As was explained in [this StackOverflow post](https://stackoverflow.com/questions/51669724/install-rt-linux-patch-for-ubuntu).

rickstaa, 2021-05-19:
Closed to the discussion in one place. New issue is https://github.com/frankaemika/franka_ros/issues/101.

rickstaa, 2021-05-20:
# Ubuntu 20.04 (kernel 5.6.19) instructions

For future reference, the steps above did not work anymore on ubuntu 20.04. The steps below seem to work when trying to compile rt kernel 5.6.19.

1.

rickstaa, 2021-05-20:
Reopened since I am going to leave this here as a reference.

gavanderhoorn, 2021-05-21:
`make deb-pkg` is the recommended way to build `.deb`s for kernel images and headers on Debian (and by extension, Ubuntu). See [Building a custom kernel from Debian kernel source](https://kernel-team.pages.debian.net/kernel-handbook/ch-common-tasks.html#s-common-building) in the Debian Kernel handbook.

Personally I wouldn't `sudo make install ..` anything on OS with a package manager, unless it can't be helped.

After patching the kernel and copying the old config to `.config`, I had to empty out `CONFIG_SYSTEM_TRUSTED_KEYS`, then `make olddefconfig`, then `make menuconfig` to configure the necessary options, then `make clean` and finally `make deb-pkg` (with `-jN` where `N` a large nr to not have to wait too long). That resulted in all the expected `.deb`s.

I ran everything in a `ubuntu:bionic` container, perhaps that helped avoid running into issues with incompatible build environments on systems-in-use.

rickstaa, 2021-05-22:
@gavanderhoorn Thanks a lot for helping me out again. I tested your steps, and with one small configuration change, I can now compile any of the real-time kernel patches [in the kernel repository](https://mirrors.edge.kernel.org/pub/linux/kernel/projects/rt/). I also see why it is the recommended way, it boots a lot faster than the real-time kernel, and it gives a warning that the Nvidia drivers are not supported for booting the real-time kernel. So I marked my steps above as non-recommended and added the new steps below.

# Ubuntu 20.04 real-time kernel build instructions (recommended way)

After the comment @gavanderhoorn, I found the right steps for building the real-time kernel using the `deb-pkg` package. These steps are very much in line with the steps given in the [franka documentation](https://frankaemika.github.io/docs/installation_linux.html#setting-up-the-real-time-kernel). Below, I marked the changed I had to make relative to the documentation to make it work no my system. Keep in mind that different steps might be needed on your system.

## Copy your current kernel config instead of creating a new config

To have a successful build, I needed to copy the configuration of the default kernel into the patched kernel. I did this using the following commands:

```bash
cd linux-5.9.1
cp -v /boot/config-$(name -r) .config
```

While doing this, you have to make sure that the kernel you are currently booted into is the same as the kernel you are trying to patch (for a good tool to install kernel version [click here](https://teejeetech.com/2019/01/20/ukuu-v19-01/)). 

## Use the make olddefconfig for setting up the config

Similar to the steps of @gavanderhoorn, I had to use the `make olddefconfig` command instead of the `make oldconfig` given in the documentation. This makes sure that the defaults are used. After this, I could use the `make menuconfig` command to set up the `PREEMPT RT` variable.

## Disable the CONFIG_DEBUG_INFO_BTF option

On my system in the `.config` file, I had to disable the `CONFIG_DEBUG_INFO_BTF` option as otherwise, it would give me an error explaining that the dwarves package is not recent enough. If you really need the debug information, you can also install version 1.17 of the dwarves package.

## Run make clean before building the kernel

On my system, I had to execute the  `make clean` command on my system before building the package on my system.

## Remove fakeroot before the build command

On my system, I could not use the `fakeroot` package to build the kernel as it would give me a `fakeroot: nested operation not yet supported` error. This resulted in the following build command `make deb-pkg -j $(nproc)`. As on my system, I have root access, so I do not need this package. For people that need to use this package, please see [franka_ros/issues/101](https://github.com/frankaemika/franka_ros/issues/101). 

## Install the kernel using the dpkg package

After all these changes, I could install the `.deb` package using the `dpkg` package.

# Extra steps for Nvidia drivers

When you are using NVidia drivers on any of your other kernels , as Nvidia doesn't offer support for these custom kernels (see [this post](https://forums.developer.nvidia.com/t/simply-impossible-to-use-an-nvidia-card-on-a-real-time-kernel-no-musician-can-use-nvidia/34947/8)), you have to switch back to nouveau drivers before you try to boot the real-time kernel. A guide to ease this process can be found [here](https://wiki.gentoo.org/wiki/Talk:Nouveau_%26_nvidia-drivers_switching0). The steps explained in that post, however, did not work on my system. Therefore, I create [this stackoverflow](https://askubuntu.com/questions/1339726/set-graphics-driver-to-nouveau-for-specific-grub-menuentry) question to find a working solution.

closed 2021-05-22 by rickstaa

carismoses, 2021-09-17:
@rickstaa I tried your (OLD) instructions with the same Ubuntu version (18.04) and kernel version (5.4.26) and got the following error. Did you encounter any errors like this?

```
caris@machine:~/linux-5.4.26$ make -j $(nproc) modules_install
cp: cannot stat './modules.builtin.modinfo': No such file or directory
Makefile:1307: recipe for target '_modinst_' failed
make: *** [_modinst_] Error 1
```

rickstaa, 2021-09-18:
@carismoses I can vaguely remember that I saw that issue. I however noticed that the issues that are encountered with the (OLD) method are dependent on your system and the settings set in `make menuconfig`. I would advise you to use https://github.com/frankaemika/libfranka/issues/62#issuecomment-846375127 or if that doesn't work https://github.com/frankaemika/libfranka/issues/62#issuecomment-844970384. If both these recipes don't work and the steps in [the documentation](https://frankaemika.github.io/docs/installation_linux.html#setting-up-the-real-time-kernel) also not work, you can also create a [new issue](https://github.com/frankaemika/libfranka/issues).

carismoses, 2021-09-20:
I was actually able to resolve it by changing one of the config options based on this [post](https://askubuntu.com/questions/1329538/compiling-the-kernel-5-11-11).
```
scripts/config --set-str SYSTEM_TRUSTED_KEYS ""
```

rickstaa, 2021-09-20:
@carismoses Great to hear you solved it and thanks for sharing your solution.

gollth, 2021-10-01:
Thank you all for solving this issue. I updated the documentation with the findings from you

medashan, 2022-01-21:
Hello,
           I am trying to set the real-time kernel. Unfortunately, I am getting errors when I try to connect to the UR5 robot externally. I think I might be missing something, and I couldn't recognize it. 
I was able to use the setup guide and also followed some of the known issues. However, I wasn't successful. 

I checked to see if the new kernel was active/installed, and it seems to be okay.
[image]

Here is the screenshot of the errors I am getting when I connect the UR5 robot. 
[image]
I knew that there were multiple issues here. 
1. Real-time capabilities
2. Calibration 
3. RTDE data pipeline

The first two issues were present before I installed the new rt kernel. The third issue came after I installed the new rt kernel. I have also generated a new calibration file; however, I am still getting the error. I thought fixing the real-time kernel issue would clear all problems. But it didn't. 

I am happy to share more information if needed. Any help would be greatly appreciated. Thank you.

gollth, 2022-01-21:
Hello @medashan, 

you do know that `libfranka` is used for Panda robots not UR5s right? I suggest you open a ticket in their Github or, better yet, buy our robot ;)

medashan, 2022-01-21:
Sorry, I didn't know that. Thanks for letting me know.

yijionglin, 2023-10-28:
Hi @rickstaa ,

Thanks for your elaborate testing on this bit!

I am trying to install a real-time kernel with Ubuntu 2204. I tried to install the RT kernel following the [document](https://frankaemika.github.io/docs/installation_linux.html#setting-up-the-real-time-kernel), however it failed in the 'make -j$(nproc) deb-pkg' process where I got some error similar to this [one](https://github.com/Frogging-Family/linux-tkg/issues/296).

Specifically, I first tried to install the RT 6.2.1 kernel with my current 6.2.0-34-generic kernel running, as the document says we'd better pick the closest version, the full commands are:

```
 $   uname -a
 $   sudo apt-get install build-essential bc curl ca-certificates gnupg2 libssl-dev lsb-release libelf-dev bison flex dwarves zstd libncurses-dev
 $   mkdir rtk
 $   cd rtk/
 $   curl -SLO https://www.kernel.org/pub/linux/kernel/v6.x/linux-6.2.1.tar.xz
 $   curl -SLO https://www.kernel.org/pub/linux/kernel/v6.x/linux-6.2.1.tar.sign
 $   curl -SLO https://mirrors.edge.kernel.org/pub/linux/kernel/projects/rt/6.2/patch-6.2-rt3.patch.xz
 $   curl -SLO https://mirrors.edge.kernel.org/pub/linux/kernel/projects/rt/6.2/patch-6.2-rt3.patch.sign
 $   xz -d *.xz
 $   gpg2 --verify linux-*.tar.sign
 $   gpg2 --verify patch-*.patch.sign
 $   tar xf linux-*.tar
 $   cd linux-*/
 $   patch -p1 < ../patch-*.patch
 $   cp -v /boot/config-$(uname -r) .config
 $   make olddefconfig
 $   make menuconfig
 $   make -j$(nproc) deb-pkg
```

This failed after around 15 mins and showed something like:

```
  CC [M]  drivers/iio/temperature/mlx90632.o
  CC [M]  drivers/iio/temperature/tmp007.o
  CC [M]  drivers/iio/temperature/tmp006.o
  CC [M]  drivers/iio/temperature/tmp117.o
  CC [M]  drivers/iio/temperature/tsys01.o
  CC [M]  drivers/iio/temperature/tsys02d.o
  LD [M]  drivers/iio/pressure/st_pressure.o
make[3]: *** [Makefile:2021: .] Error 2
make[2]: *** [debian/rules:7: build-arch] Error 2
dpkg-buildpackage: error: debian/rules binary subprocess returned exit status 2
make[1]: *** [scripts/Makefile.package:90: deb-pkg] Error 2
make: *** [Makefile:1665: deb-pkg] Error 2
```

Later, I also tried first installing the 6.2.1 kernel with a tool called Ukuu Mainline following [here](https://www.groovypost.com/howto/how-to-downgrade-the-kernel-in-ubuntu/), and then rebooted and selected the 6.2.1 kernel (used "uname -r" to confirm it). After that, I ran exactly the same steps from the Franka [document](https://frankaemika.github.io/docs/installation_linux.html#setting-up-the-real-time-kernel). And it still met the same error.

Could you please give me some suggestions on this bit? 

Many thanks,
Bourne

yijionglin, 2023-10-28:
> Hi @rickstaa ,
> 
> Thanks for your elaborate testing on this bit!
> 
> I am trying to install a real-time kernel with Ubuntu 2204. I tried to install the RT kernel following the [document](https://frankaemika.github.io/docs/installation_linux.html#setting-up-the-real-time-kernel), however it failed in the 'make -j$(nproc) deb-pkg' process where I got some error similar to this [one](https://github.com/Frogging-Family/linux-tkg/issues/296).
> 
> Specifically, I first tried to install the RT 6.2.1 kernel with my current 6.2.0-34-generic kernel running, as the document says we'd better pick the closest version, the full commands are:
> 
> ```
>  $   uname -a
>  $   sudo apt-get install build-essential bc curl ca-certificates gnupg2 libssl-dev lsb-release libelf-dev bison flex dwarves zstd libncurses-dev
>  $   mkdir rtk
>  $   cd rtk/
>  $   curl -SLO https://www.kernel.org/pub/linux/kernel/v6.x/linux-6.2.1.tar.xz
>  $   curl -SLO https://www.kernel.org/pub/linux/kernel/v6.x/linux-6.2.1.tar.sign
>  $   curl -SLO https://mirrors.edge.kernel.org/pub/linux/kernel/projects/rt/6.2/patch-6.2-rt3.patch.xz
>  $   curl -SLO https://mirrors.edge.kernel.org/pub/linux/kernel/projects/rt/6.2/patch-6.2-rt3.patch.sign
>  $   xz -d *.xz
>  $   gpg2 --verify linux-*.tar.sign
>  $   gpg2 --verify patch-*.patch.sign
>  $   tar xf linux-*.tar
>  $   cd linux-*/
>  $   patch -p1 < ../patch-*.patch
>  $   cp -v /boot/config-$(uname -r) .config
>  $   make olddefconfig
>  $   make menuconfig
>  $   make -j$(nproc) deb-pkg
> ```
> 
> This failed after around 15 mins and showed something like:
> 
> ```
> ...
>   CC [M]  drivers/comedi/drivers/ni_labpc_common.o
>   CC [M]  drivers/comedi/drivers/ni_labpc_isadma.o
>   LD [M]  drivers/iio/pressure/st_pressure.o
>   CC [M]  drivers/comedi/drivers/comedi_8255.o
>   CC [M]  drivers/infiniband/hw/hfi1/debugfs.o
>   CC [M]  drivers/comedi/drivers/8255.o
>   CC [M]  drivers/comedi/drivers/amplc_dio200_common.o
>   CC [M]  drivers/comedi/drivers/amplc_pc236_common.o
>   CC [M]  drivers/comedi/drivers/das08.o
>   LD [M]  drivers/comedi/drivers/ni_routing.o
>   LD [M]  drivers/infiniband/hw/hfi1/hfi1.o
> make[3]: *** [Makefile:2021: .] Error 2
> make[2]: *** [debian/rules:7: build-arch] Error 2
> dpkg-buildpackage: error: debian/rules binary subprocess returned exit status 2
> make[1]: *** [scripts/Makefile.package:90: deb-pkg] Error 2
> make: *** [Makefile:1665: deb-pkg] Error 2
> ```
> 
> Later, I also tried first installing the 6.2.1 kernel with a tool called Ukuu Mainline following [here](https://www.groovypost.com/howto/how-to-downgrade-the-kernel-in-ubuntu/), and then rebooted and selected the 6.2.1 kernel (used "uname -r" to confirm it). After that, I ran exactly the same steps from the Franka [document](https://frankaemika.github.io/docs/installation_linux.html#setting-up-the-real-time-kernel). And it still met the same error.
> 
> Could you please give me some suggestions on this bit?
> 
> Many thanks, Bourne

By the way, when I tried to run the again after the previous error showed up, this time it didn't do the build that long and stop within 10 seconds with this error:

```
(franka) bourne@bourne:~/dev/franka/rtk/linux-6.2.1$ make -j$(nproc) deb-pkg
make clean
  CLEAN   arch/x86/lib
  CLEAN   crypto/asymmetric_keys
  CLEAN   arch/x86/crypto
  CLEAN   init
  CLEAN   arch/x86/kvm
  CLEAN   arch/x86/entry/vdso
  CLEAN   drivers/accessibility/speakup
  CLEAN   kernel/debug/kdb
[... 39 lines cut ...]
dpkg-source: warning: source directory 'linux-6.2.1' is not <sourcepackage>-<upstreamversion> 'linux-upstream-6.2.1-rt3'
dpkg-source: warning: .orig directory name linux-6.2.1.orig is not <package>-<upstreamversion> (wanted linux-upstream-6.2.1-rt3.orig)
dpkg-source: info: building linux-upstream using existing linux-upstream_6.2.1-rt3.orig.tar.gz
dpkg-source: info: building linux-upstream in linux-upstream_6.2.1-rt3-2.diff.gz
dpkg-source: error: cannot represent change to vmlinux-gdb.py:
dpkg-source: error:   new version is symlink to /home/bourne/dev/franka/rtk/linux-6.2.1/scripts/gdb/vmlinux-gdb.py
dpkg-source: error:   old version is nonexistent
dpkg-source: warning: ignoring deletion of file .scmversion
dpkg-source: warning: the diff modifies the following upstream files: 
 .clang-format
 .cocciconfig
 .config.old
 .get_maintainer.ignore
 .mailmap
 .rustfmt.toml
 .version
 CREDITS
 MAINTAINERS
 README
dpkg-source: info: use the '3.0 (quilt)' format to have separate and documented changes to upstream files, see dpkg-source(1)
dpkg-source: error: unrepresentable changes to source
dpkg-buildpackage: error: dpkg-source -i.git -b . subprocess returned exit status 1
make[1]: *** [scripts/Makefile.package:90: deb-pkg] Error 1
make: *** [Makefile:1665: deb-pkg] Error 2
```

yijionglin, 2023-10-28:
> > Hi @rickstaa ,
> > Thanks for your elaborate testing on this bit!
> > I am trying to install a real-time kernel with Ubuntu 2204. I tried to install the RT kernel following the [document](https://frankaemika.github.io/docs/installation_linux.html#setting-up-the-real-time-kernel), however it failed in the 'make -j$(nproc) deb-pkg' process where I got some error similar to this [one](https://github.com/Frogging-Family/linux-tkg/issues/296).
> > Specifically, I first tried to install the RT 6.2.1 kernel with my current 6.2.0-34-generic kernel running, as the document says we'd better pick the closest version, the full commands are:
> > ```
> >  $   uname -a
> >  $   sudo apt-get install build-essential bc curl ca-certificates gnupg2 libssl-dev lsb-release libelf-dev bison flex dwarves zstd libncurses-dev
> >  $   mkdir rtk
> >  $   cd rtk/
> >  $   curl -SLO https://www.kernel.org/pub/linux/kernel/v6.x/linux-6.2.1.tar.xz
> >  $   curl -SLO https://www.kernel.org/pub/linux/kernel/v6.x/linux-6.2.1.tar.sign
> >  $   curl -SLO https://mirrors.edge.kernel.org/pub/linux/kernel/projects/rt/6.2/patch-6.2-rt3.patch.xz
> >  $   curl -SLO https://mirrors.edge.kernel.org/pub/linux/kernel/projects/rt/6.2/patch-6.2-rt3.patch.sign
> >  $   xz -d *.xz
> >  $   gpg2 --verify linux-*.tar.sign
> >  $   gpg2 --verify patch-*.patch.sign
> >  $   tar xf linux-*.tar
> >  $   cd linux-*/
> >  $   patch -p1 < ../patch-*.patch
> >  $   cp -v /boot/config-$(uname -r) .config
> >  $   make olddefconfig
> >  $   make menuconfig
> >  $   make -j$(nproc) deb-pkg
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > This failed after around 15 mins and showed something like:
> > ```
> > ...
> >   CC [M]  drivers/comedi/drivers/ni_labpc_common.o
> >   CC [M]  drivers/comedi/drivers/ni_labpc_isadma.o
> >   LD [M]  drivers/iio/pressure/st_pressure.o
> >   CC [M]  drivers/comedi/drivers/comedi_8255.o
> >   CC [M]  drivers/infiniband/hw/hfi1/debugfs.o
> >   CC [M]  drivers/comedi/drivers/8255.o
> >   CC [M]  drivers/comedi/drivers/amplc_dio200_common.o
> >   CC [M]  drivers/comedi/drivers/amplc_pc236_common.o
> >   CC [M]  drivers/comedi/drivers/das08.o
> >   LD [M]  drivers/comedi/drivers/ni_routing.o
> >   LD [M]  drivers/infiniband/hw/hfi1/hfi1.o
> > make[3]: *** [Makefile:2021: .] Error 2
> > make[2]: *** [debian/rules:7: build-arch] Error 2
> > dpkg-buildpackage: error: debian/rules binary subprocess returned exit status 2
> > make[1]: *** [scripts/Makefile.package:90: deb-pkg] Error 2
> > make: *** [Makefile:1665: deb-pkg] Error 2
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > Later, I also tried first installing the 6.2.1 kernel with a tool called Ukuu Mainline following [here](https://www.groovypost.com/howto/how-to-downgrade-the-kernel-in-ubuntu/), and then rebooted and selected the 6.2.1 kernel (used "uname -r" to confirm it). After that, I ran exactly the same steps from the Franka [document](https://frankaemika.github.io/docs/installation_linux.html#setting-up-the-real-time-kernel). And it still met the same error.
> > Could you please give me some suggestions on this bit?
> > Many thanks, Bourne
> 
[... 82 lines cut ...]
Before `sudo make`, you need to modify these lines in the .config file that copied from /boot/config-xxxx-generic:

`CONFIG_SYSTEM_TRUSTED_KEYS="debian/canonical-certs.pem"`

Change this to this:

`CONFIG_SYSTEM_TRUSTED_KEYS=""`

And 

`CONFIG_SYSTEM_REVOCATION_KEYS="debian/canonical-revoked-certs.pem"`

to 

`CONFIG_SYSTEM_REVOCATION_KEYS=""`

(The latter key has been added to the default Canonical kernel configuration since some time.)

As I disable the keys, instead of `make -j$(nproc) deb-pkg`, I just used `make -j$(nproc)` omitting deb-pkg. Otherwise you would meet some errors related to the dpkgs-source.

Reference:

1. https://www.acontis.com/en/building-a-real-time-linux-kernel-in-ubuntu-preemptrt.html
2. https://askubuntu.com/questions/1329538/compiling-the-kernel-5-11-11
3. My genius fellow.

rickstaa, 2023-10-30:
> > > Hi @rickstaa ,
> > > Thanks for your elaborate testing on this bit!
> > > I am trying to install a real-time kernel with Ubuntu 2204. I tried to install the RT kernel following the [document](https://frankaemika.github.io/docs/installation_linux.html#setting-up-the-real-time-kernel), however it failed in the 'make -j$(nproc) deb-pkg' process where I got some error similar to this [one](https://github.com/Frogging-Family/linux-tkg/issues/296).
> > > Specifically, I first tried to install the RT 6.2.1 kernel with my current 6.2.0-34-generic kernel running, as the document says we'd better pick the closest version, the full commands are:
> > > ```
> > >  $   uname -a
> > >  $   sudo apt-get install build-essential bc curl ca-certificates gnupg2 libssl-dev lsb-release libelf-dev bison flex dwarves zstd libncurses-dev
> > >  $   mkdir rtk
> > >  $   cd rtk/
> > >  $   curl -SLO https://www.kernel.org/pub/linux/kernel/v6.x/linux-6.2.1.tar.xz
> > >  $   curl -SLO https://www.kernel.org/pub/linux/kernel/v6.x/linux-6.2.1.tar.sign
> > >  $   curl -SLO https://mirrors.edge.kernel.org/pub/linux/kernel/projects/rt/6.2/patch-6.2-rt3.patch.xz
> > >  $   curl -SLO https://mirrors.edge.kernel.org/pub/linux/kernel/projects/rt/6.2/patch-6.2-rt3.patch.sign
> > >  $   xz -d *.xz
> > >  $   gpg2 --verify linux-*.tar.sign
> > >  $   gpg2 --verify patch-*.patch.sign
> > >  $   tar xf linux-*.tar
> > >  $   cd linux-*/
> > >  $   patch -p1 < ../patch-*.patch
> > >  $   cp -v /boot/config-$(uname -r) .config
> > >  $   make olddefconfig
> > >  $   make menuconfig
> > >  $   make -j$(nproc) deb-pkg
> > > ```
> > > 
> > > 
> > >     
> > >       
> > >     
> > > 
> > >       
> > >     
> > > 
> > >     
> > >   
> > > This failed after around 15 mins and showed something like:
> > > ```
> > > ...
> > >   CC [M]  drivers/comedi/drivers/ni_labpc_common.o
> > >   CC [M]  drivers/comedi/drivers/ni_labpc_isadma.o
> > >   LD [M]  drivers/iio/pressure/st_pressure.o
> > >   CC [M]  drivers/comedi/drivers/comedi_8255.o
> > >   CC [M]  drivers/infiniband/hw/hfi1/debugfs.o
> > >   CC [M]  drivers/comedi/drivers/8255.o
> > >   CC [M]  drivers/comedi/drivers/amplc_dio200_common.o
> > >   CC [M]  drivers/comedi/drivers/amplc_pc236_common.o
> > >   CC [M]  drivers/comedi/drivers/das08.o
> > >   LD [M]  drivers/comedi/drivers/ni_routing.o
> > >   LD [M]  drivers/infiniband/hw/hfi1/hfi1.o
> > > make[3]: *** [Makefile:2021: .] Error 2
> > > make[2]: *** [debian/rules:7: build-arch] Error 2
> > > dpkg-buildpackage: error: debian/rules binary subprocess returned exit status 2
> > > make[1]: *** [scripts/Makefile.package:90: deb-pkg] Error 2
> > > make: *** [Makefile:1665: deb-pkg] Error 2
> > > ```
> > > 
> > > 
> > >     
> > >       
> > >     
> > > 
> > >       
> > >     
> > > 
> > >     
> > >   
> > > Later, I also tried first installing the 6.2.1 kernel with a tool called Ukuu Mainline following [here](https://www.groovypost.com/howto/how-to-downgrade-the-kernel-in-ubuntu/), and then rebooted and selected the 6.2.1 kernel (used "uname -r" to confirm it). After that, I ran exactly the same steps from the Franka [document](https://frankaemika.github.io/docs/installation_linux.html#setting-up-the-real-time-kernel). And it still met the same error.
> > > Could you please give me some suggestions on this bit?
> > > Many thanks, Bourne
> > 
[... 86 lines cut ...]
> Change this to this:
> 
> `CONFIG_SYSTEM_TRUSTED_KEYS=""`
> 
> And
> 
> `CONFIG_SYSTEM_REVOCATION_KEYS="debian/canonical-revoked-certs.pem"`
> 
> to
> 
> `CONFIG_SYSTEM_REVOCATION_KEYS=""`
> 
> (The latter key has been added to the default Canonical kernel configuration since some time.)
> 
> As we disable the keys, instead of `make -j$(nproc) deb-pkg`, we just used `make -j$(nproc)` omitting deb-pkg. Otherwise we would meet some errors related to the dpkgs-source.
> 
> FYI, my current kernel is 6.2.0-34-generic and the RT kernel I installed is 6.2.1-rt3. I tried kernel 5.15.96-rt61 which also worked. My experience is: when you do `make -j$(nproc) deb-pkg`, if there is any error prompt, it probably does not show the real error caused by debian package or configuration. Might be wrong with this bit.
> 
> Reference:
> 
> 1. https://www.acontis.com/en/building-a-real-time-linux-kernel-in-ubuntu-preemptrt.html
> 2. https://askubuntu.com/questions/1329538/compiling-the-kernel-5-11-11
> 3. My genius fellow.

It's good to learn that you cracked the code on your issue! Thanks for sharing the solution here so other users can use it in the future! 🌟