frankarobotics/libfranka #63: Other way to unlock joints than from desk

liborw, 2020-05-26:
Is it possible to unlock joints programmatically?

Hymwgk, 2020-08-06:
Same question

Hymwgk, 2020-08-06:
Hi, I got this ,but I haven't try it yet
[https://github.com/ib101/DVK/blob/master/Code/DVK.py](url)

eugenegalaxy, 2020-09-27:
> Hi, I got this ,but I haven't try it yet
> [https://github.com/ib101/DVK/blob/master/Code/DVK.py](url)

Hey, I have tried this code (unmodified) and it worked! Using Python 3.6, Ubuntu 18.04.

jk-ethz, 2022-11-08:
My solution to this
https://github.com/jk-ethz/franka-lock-unlock

AndreasKuhner, 2024-09-26:
Hi all,
while not providing an official way of locking/unlocking joints, there seems to be workarounds... :)

Cheers,
Andreas

P.S. If you have further input, please open another issue.

closed 2024-09-26 by AndreasKuhner