frankarobotics/libfranka #65: Include headers missing

nash169, 2020-06-03:
Hi,
I've noticed that a couple of include headers are missing and this might break the compilation some time.

In include/franka/control_tools.h missing #include <string>
In include/franka/control_types.h missing #include <stdexcept>

otherwiseopen, 2021-05-13:
Hi,
do you plan to update the 0.8.* release and the [libfranka-release repo](https://github.com/frankaemika/libfranka-release) with this fix? Currently, it is not possible building a Franka-enabled ROS from source using the [recommended process](http://wiki.ros.org/noetic/Installation/Source) without some awkward workarounds (at least on Fedora 33/34), which is unfortunate, since the problem has been fixed for quite a while.

AndreasKuhner, 2024-09-26:
Hi @nash169 ,
I think the issue is resolved by now... if not, please open again. Thank you!

Cheers,
Andreas

closed 2024-09-26 by AndreasKuhner