frankarobotics/libfranka #70: build fail on run_all_tests

M-A-Robson, 2020-06-24:
cannot build libfranka from source to work with the latest hardware update.

I follow the instructions here: https://frankaemika.github.io/docs/installation_linux.html

get to:
cmake --build .

but receive the following errors about 50% of the way through build:

Linking CXX executable run_all_tests
CMakeFiles/run_all_tests.dir/mock_server.cpp.o: In function `MockServer<RobotTypes>::serverThread()':
mock_server.cpp:(.text._ZN10MockServerI10RobotTypesE12serverThreadEv[_ZN10MockServerI10RobotTypesE12serverThreadEv]+0x84): undefined reference to `Poco::Net::SocketAddress::SocketAddress(std::string const&, unsigned short)'
mock_server.cpp:(.text._ZN10MockServerI10RobotTypesE12serverThreadEv[_ZN10MockServerI10RobotTypesE12serverThreadEv]+0x443): undefined reference to `Poco::Net::SocketAddress::SocketAddress(std::string const&, unsigned short)'
CMakeFiles/run_all_tests.dir/mock_server.cpp.o: In function `MockServer<GripperTypes>::serverThread()':
mock_server.cpp:(.text._ZN10MockServerI12GripperTypesE12serverThreadEv[_ZN10MockServerI12GripperTypesE12serverThreadEv]+0x87): undefined reference to `Poco::Net::SocketAddress::SocketAddress(std::string const&, unsigned short)'
mock_server.cpp:(.text._ZN10MockServerI12GripperTypesE12serverThreadEv[_ZN10MockServerI12GripperTypesE12serverThreadEv]+0x439): undefined reference to `Poco::Net::SocketAddress::SocketAddress(std::string const&, unsigned short)'
CMakeFiles/run_all_tests.dir/mock_server.cpp.o: In function `MockServer<VacuumGripperTypes>::serverThread()':
mock_server.cpp:(.text._ZN10MockServerI18VacuumGripperTypesE12serverThreadEv[_ZN10MockServerI18VacuumGripperTypesE12serverThreadEv]+0x87): undefined reference to `Poco::Net::SocketAddress::SocketAddress(std::string const&, unsigned short)'
CMakeFiles/run_all_tests.dir/mock_server.cpp.o:mock_server.cpp:(.text._ZN10MockServerI18VacuumGripperTypesE12serverThreadEv[_ZN10MockServerI18VacuumGripperTypesE12serverThreadEv]+0x439): more undefined references to `Poco::Net::SocketAddress::SocketAddress(std::string const&, unsigned short)' follow
../libfranka.so.0.8.0: undefined reference to `Poco::SharedLibrary::suffix()'
../libfranka.so.0.8.0: undefined reference to `Poco::File::File(std::string const&)'
../libfranka.so.0.8.0: undefined reference to `Poco::TemporaryFile::registerForDeletion(std::string const&)'
../libfranka.so.0.8.0: undefined reference to `Poco::SharedLibrary::getSymbol(std::string const&)'
../libfranka.so.0.8.0: undefined reference to `Poco::SharedLibrary::load(std::string const&)'
../libfranka.so.0.8.0: undefined reference to `Poco::TemporaryFile::tempName()'
collect2: error: ld returned 1 exit status
test/CMakeFiles/run_all_tests.dir/build.make:569: recipe for target 'test/run_all_tests' failed
make[2]: *** [test/run_all_tests] Error 1
CMakeFiles/Makefile2:150: recipe for target 'test/CMakeFiles/run_all_tests.dir/all' failed
make[1]: *** [test/CMakeFiles/run_all_tests.dir/all] Error 2
Makefile:160: recipe for target 'all' failed
make: *** [all] Error 2

I've tried updating my compiler, and manually installing POCO libs but no luck. install works from apt-get ros-kinetic-libfranka but this is version 0.7.0 which is not compatible with the latest update on the robot and so is throwing errors when trying to connect through ROS.

sgabl, 2020-06-24:
Hi,

in general, you can deactivate building the tests. (`cmake -DCMAKE_BUILD_TYPE=Release -DBUILD_TESTS=OFF ..`) However, this is not the issue you have. Could you check if Poco is found (e.g. use `ccmake .` in your build folder).

Travis has no problem  [building `libfranka 0.8.0` on 16.04](https://travis-ci.org/github/frankaemika/libfranka/jobs/699259668).

M-A-Robson, 2020-06-24:
I found that i was able to make the build work by building and installing POCO from source after reading that you may want to compile POCO and your c++ project using the same compiler.

I followed the instructions here: https://pocoproject.org/download.html

closed 2020-06-24 by M-A-Robson