frankarobotics/libfranka #74: Feature Request: setSelfCollisionBehavior()

ndehio, 2020-07-01:
We can already set thresholds for joint/cartesian collisions via the `setCollisionBehavior()` function.
Could you please also provide a method to modify the thresholds for self-collisions?
Thanks

Stakkay, 2021-02-24:
The [setCollisionBehavior()](https://frankaemika.github.io/libfranka/classfranka_1_1Robot.html#a168e1214ac36d74ac64f894332b84534) function let's the user set force and torque thresholds to detect contacts and collisions. Self-collisions on the other hand happen when the geometry of the robot intersects with itself (i.e. one part of the robot gets too close to another part of the robot). What thresholds do you want to set for self-collisions?

AndreasKuhner, 2024-09-26:
Hi @ndehio 
as @Stakkay mentioned, setting thresholds for self-collisions (aka the robot hits itself), is more like a thing of 'please don't do'. Hence, the collision volumes are chosen to prevent any self-harm (and not configurable).

Cheers,
Andreas

closed 2024-09-26 by AndreasKuhner