frankarobotics/libfranka #77: libfranka: unable to set realtime scheduling: Operation not permitted

YinjieLin, 2020-08-03:
Hello guys!
I am using Franka with my computer as following:
[image]
Also, I install the realtime kernel followed by the instructions of FCI documents, and after the installation, I used the rt-test to test if the real-time kernel was worked well,
[image]
All the results seem OK, however, when I restart and enter in the kernel 4.14.12_rt10, errors happened
[image]
and, the Graphic driver and wireless driver did not work well.
When I start the roslauch with Franka example controllers, errors shown as "libfranka: unable to set realtime scheduling: Operation not permitted"
Detailed:
[image]
[errors.txt](https://github.com/frankaemika/libfranka/files/5014908/errors.txt)

I have tried to reinstall the real-time kernel many times, with 4.14.12 or other 4.14 versions, the results were the same. Is the problem because of the i9 9900k cup? Since no problem happened on my laptop(i7 8650u).
Are there any solutions to this problem?

EricCousineau-TRI, 2020-12-08:
(FYI Just a fellow user here)

If it's not suuuper important that you run realtime, then you can use `realtime_config = RealtimeConfig::kIgnore`:
https://frankaemika.github.io/libfranka/classfranka_1_1Robot.html#ae63bc19390df3d54f3a270814df35eb6

Before delving into anything ROS-laden, you may want to try one of the direct examples, e.g.:
https://github.com/frankaemika/libfranka/blob/f1f46fb008a37eb0d1dba00c971ff7e5a7bfbfd3/examples/generate_joint_position_motion.cpp

Then mayhaps try something like `cd build && sudo ./examples/generate_joint_position_motion <robot-address>` to see if it's a permissions issue.

If so, then maybe see if you can configure `rtprio` to allow priority changes. (A colleague had set this up on our side, I dunno the specifics):
https://serverfault.com/questions/487602/linux-etc-security-limits-conf-explanation

DaHornung, 2021-03-11:
@YinjieLin I faced the same issue on Ubuntu 18.04 and 5.4.102-rt53 #1 SMP PREEMPT_RT kernel.
Even with the libfranka examples. ( I don't think it does not matter if you launch a file in ROS or use an example code, as this is probably a systems error.
I also consulted the steps in:
https://frankaemika.github.io/docs/troubleshooting.html
and set the cpu performance up to performance and verified.

Jitter test gave:
--- 172.16.0.2 ping statistics ---
10000 packets transmitted, 10000 received, 0% packet loss, time 9998ms
rtt min/avg/max/mdev = 0.080/0.185/0.354/0.037 ms

And I double-checked that user is within the realtime group with:
`getent group`
Result `realtime:x:1001:userName`
I also logged off and on and also rebooted.

DaHornung, 2021-03-11:
My bad, I somehow commented the configurations in the `limits.config`:

@realtime soft rtprio 99
@realtime soft priority 99
@realtime soft memlock 102400
@realtime hard rtprio 99
@realtime hard priority 99
@realtime hard memlock 102400

That resolved the error.

Also I did not use fakeroot for makeing the kernel. Just: `make -j4 deb-pkg`

hussienali321, 2021-04-15:
How you commented on the limits.conf ??because when I add them to the file I can't save the file

DaHornung, 2021-04-15:
Ah yeah, I was playing a bit.
I tried saving it under a different (temporary) file name. The command prompt opens the same questions again then I typed in the original file name. I re-opened the files to double-check if the changes were applied.

Regards

hussienali321, 2021-04-15:
Can you please give me more information to understand you? 
Thanks for your help

AndreasKuhner, 2024-09-26:
Hi @hussienali321 ,
could you get it to work? If not, don't hesitate to re-open the issue here.

Regarding the file which was edited - I guess you would need to open it as 'sudo' in order to be able to save it. 

Cheers,
Andreas

closed 2024-09-26 by AndreasKuhner