frankarobotics/libfranka #78: libfranka: Move command aborted: motion aborted by reflex! ["cartesian_motion_generator_joint_acceleration_discontinuity"]

Africaggy, 2020-08-04:
when I have download libfranka,I use "cmake.."and"make"in libfranka/example. Then I got some executable files, I run some files perfect.But I run "generate_cartesian_pose_motion"wrong,apppeared the above probelm. Have anyone this problem?Or have anyone can solve problem and have a new pose_motion

Africaggy, 2020-08-07:
I have solved it.Thanks guys

closed 2020-10-21 by sgabl

michtesar, 2020-10-28:
Hey @Africaggy how did you solved it?
This is what I get when I send joints states. After some I get the same error:
```
libfranka: Move command aborted: motion aborted by reflex! ["cartesian_motion_generator_joint_acceleration_discontinuity"]
```

whyy812, 2021-07-05:
Hey, @Africaggy 
We are struggling with the similar issues, too. Now that you have  solved the similar issues, could you please share the solutions ?
Best regards.

Africaggy, 2021-07-05:
I think you can smooth its motion。But you need use Quintic spline fitting

发自我的iPhone

whyy812, 2021-07-05:
I will try. Thank you @Africaggy.

Africaggy, 2021-07-05:
You are welcome

发自我的iPhone

undefinedcodezhong, 2023-05-11:
Hello, I want to know if there is an instruction or function to recover the movable state by code way when the robot is in the reflex state.
Anyone can help?  Thanks.

marcbone, 2023-05-11:
You can use the [automaticErrorRecovery](https://github.com/frankaemika/libfranka/blob/f3b8d775a9c847cab32684c8a316f67867761674/include/franka/robot.h#LL678C8-L678C30) method for that

undefinedcodezhong, 2023-05-11:
Thank you, it helps!