frankarobotics/libfranka #81: Reading RobotState while a controller is running

a-z-e-r-i-l-a, 2020-08-24:
I have an implementation of a controller which uses the [control method](http://docs.ros.org/kinetic/api/franka_hw/html/classfranka__hw_1_1FrankaHW.html#a75bd1438ffbfa54ce3ff6a4364df1cc0) of franka_hw to move the robot. While this controller is running I wrote another node that connects to the robot by    ` franka::Robot robot(robot_ip);` and uses `robot.readOnce()` to see the current state of the robot. Howeve, I noticed that I can't run both of these codes at the same time and a `franka::NetworkException´ and ´libfranka: UDP receive: Timeout` is raised if I do so.

Is there any way however to read the robot's state while a controller is already running?

LinJianjie, 2023-03-15:
ask the same question

marcbone, 2023-03-15:
As you said the readOnce cannot be called while controlling the robot. When a controller is running you get the RobotState in the control callback. You can pass it to another thread by protecting it with a mutex. See the [joint_impedance_example](https://github.com/frankaemika/libfranka/blob/master/examples/joint_impedance_control.cpp)

closed 2023-03-15 by marcbone