frankarobotics/libfranka #83: Can't build libfranka in Windows

dborro, 2020-11-01:
I tried to compile and build libfranka in Windows 10 using Visual Studio 2019 but I have the following error:

Description: 'Poco::Foundation-NOTFOUND', needed by 'franka.dll', missing and no known rule to make it
Project: C:\src\vcpkg\libfranka\out\build\x64-Debug\libfranka
File: C:\src\vcpkg\libfranka\out\build\x64-Debug\ninja

I followed all the instructions for windows (https://frankaemika.github.io/docs/installation_windows.html) but in the last step (building), it seems the project can't locate the path to poco. I also tried to put directly dll files into the chosen build root (as the instructions say) but with the same error.

THELUOHAO, 2021-01-08:
it is just the problem of the wrong path. please ensure the right path can be found by a compiler. but how to combine the libfranka and the frame of MFC project. it really fuzzy me. Maybe we could have a discussion.

avadhutmali999, 2022-07-26:
Facing a same issue. Please guide if someone has solved it already.
Need it on URGENT basis. Please HELP.

avadhutmali999, 2022-08-02:
Solved the issue partially ,check whether it works for you.

1. Install vcpkg using administrator CMD prompt only.
2. Install eigen3 and poco as installed earlier.
3. Enter this line in command prompt "vcpkg integrate install" .
4. Open CMakelists.txt file in note pad, and set "BUILD_test" to OFF.
5. Save the Cmakelists.txt file.
6. Follow the remaining instructions for Visual Studio as per franka documentation.

Still after doing all of this, I was able to get franka.dll file after building libfranka but not franka.exe .

fcc315, 2022-09-06:
> Solved the issue partially ,check whether it works for you.
> 
> 1. Install vcpkg using **administrator** CMD prompt only.
> 2. Install eigen3 and poco as installed earlier.
> 3. Enter this line in command prompt "vcpkg integrate install" .
> 4. Open CMakelists.txt file in note pad, and set "BUILD_test" to OFF.
> 5. Save the Cmakelists.txt file.
> 6. Follow the remaining instructions for Visual Studio as per franka documentation and build libfranka accordingly.
> 7. Franka.dll and pocoNetd.dll file will be placed in libfranka\out\build..
> 8. Copy paste these files to C:\windows\System32.
> 
> Now You'll be able to run the examples.

Your answer is very helpful, thanks! 
Following your steps, I still have an error. There is some lines relate to Debian package in CMakeLists.txt line 183-191. Do you have this problem and how to solve this?

avadhutmali999, 2022-09-06:
> Your answer is very helpful, thanks! Following your steps, I still have an error. There is some lines relate to Debian package in CMakeLists.txt line 183-191. Do you have this problem and how to [...]

Can you elaborate the error with snapshots?

fcc315, 2022-09-21:
> Hey I attached a snapshot. These lines can be found in CMakeLists.txt in libfranka folder. I wonder what these are used for. I comment out them and compiled successfully.

[image]

fcc315, 2022-09-27:
> Not sure what it means as I am working only on windows. Nevertheless if you are able to get the BUILD its totally fine to comment out these. Also I need some help with building my own code for PANDA Robot, getting a few issues using CMake, could you please help me out?

sure. you can list your issues here and let's see. I am also working on windows, let's see what I can help

Zahras17, 2023-02-27:
Hi, 
I want to start programming the Panda robot. Since you already have installed libfranka on windows, I wanted to make sure if this method works appropriately or if I should go for the Linux-based approach. Any advice based on your experience could be helpful. Thanks in advance for your response.

avadhutmali999, 2023-02-27:
Yes, the approach i have followed works perfectly. Please make sure you
have full access to internet. Using proxy can cause issues sometimes.
Connect to me if you face any issues.

Zahras17, 2023-04-03:
**Couldn't find Poco**

I've tried to follow the installation steps on windows (https://www.franka.cn/FCI/installation_windows.html). When I installed Poco I received the successful installation message: 

`Computing installation plan...
The following packages are already installed:
    poco[core]:x86-windows -> 1.12.4#4
poco:x86-windows is already installed`

However, when I open the CmakLists in VS, I get this error and can not generate the CMakeSettings.json: 

`Severity	Code	Description	Project	File	Line	Suppression State
Error		CMake Error at C:/Program Files/Microsoft Visual Studio/2022/Community/Common7/IDE/CommonExtensions/Microsoft/CMake/CMake/share/cmake-3.25/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find Poco (missing: Poco_INCLUDE_DIR Poco_LIBRARIES Net
  Foundation)		C:/Program Files/Microsoft Visual Studio/2022/Community/Common7/IDE/CommonExtensions/Microsoft/CMake/CMake/share/cmake-3.25/Modules/FindPackageHandleStandardArgs.cmake	230	
` 

I tried to use the Developer Command Prompt for VS to build libfranka as well but I got this error: 

`-- The CXX compiler identification is MSVC 19.35.32216.1
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Program Files/Microsoft Visual Studio/2022/Community/VC/Tools/MSVC/14.35.32215/bin/Hostx86/x86/cl.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMake Error at C:/Program Files/Microsoft Visual Studio/2022/Community/Common7/IDE/CommonExtensions/Microsoft/CMake/CMake/share/cmake-3.25/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find Poco (missing: Poco_INCLUDE_DIR Poco_LIBRARIES Net
  Foundation)
Call Stack (most recent call first):
  C:/Program Files/Microsoft Visual Studio/2022/Community/Common7/IDE/CommonExtensions/Microsoft/CMake/CMake/share/cmake-3.25/Modules/FindPackageHandleStandardArgs.cmake:600 (_FPHSA_FAILURE_MESSAGE)
  cmake/FindPoco.cmake:27 (find_package_handle_standard_args)
  CMakeLists.txt:28 (find_package)

-- Configuring incomplete, errors occurred!
See also "C:/Users/zsafa/Desktop/Zahra/vcpkg/libfranka/build/CMakeFiles/CMakeOutput.log".`

Any help is appreciated!

avadhutmali999, 2023-04-04:
Try following these steps once and let me know if it solves the issue.

https://github.com/frankaemika/libfranka/issues/102#issuecomment-1492554875

Zahras17, 2023-04-05:
Many thanks for your response. I tried the steps in the link and also installed again the vcpkg, Eigen3, and Poco as you mentioned.  Now I receive this error: 

CMake Error at C:/Program Files/Microsoft Visual Studio/2022/Community/Common7/IDE/CommonExtensions/Microsoft/CMake/CMake/share/cmake-3.25/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find Poco (missing: Poco_INCLUDE_DIR Poco_LIBRARIES Net Foundation)

I would be grateful for any support you can provide.

avadhutmali999, 2023-04-06:
check whether there are any folders visible in following directory  = path to vcpkg\installed\x64-windows\share

Zahras17, 2023-04-06:
Thanks for your fast reply. There is a folder for Poco and I entered exactly this directory in the code line:
`cmake -DPoco_DIR=/Path/To/Poco/CMake/Config -DEigen3_DIR=/Path/To/Eigen/CMake/Config -G Ninja ..`

[image]

avadhutmali999, 2023-04-06:
Is the issue resolved now?
also build in release mode as follows
`cmake -DPoco_DIR=/Path/To/Poco/CMake/Config -DEigen3_DIR=/Path/To/Eigen/CMake/Config -DCMAKE_BUILD_TYPE=Release -G Ninja .. `

yixing312, 2024-09-25:
I've tried to follow the installation steps on windows (https://www.franka.cn/FCI/installation_windows.html) and used  the Developer Command Prompt。But was told with following code 
```
cmake -DCMAKE_TOOLCHAIN_FILE=D:/environment/vcpkg/vcpkg/scripts/buildsystems/vcpkg.cmake -G Ninja ..
CMake Deprecation Warning at CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 3.5 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

CMake Error at D:/environment/vcpkg/vcpkg/scripts/buildsystems/vcpkg.cmake:859 (_find_package):
  By not providing "Findpinocchio.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "pinocchio", but CMake did not find one.

  Could not find a package configuration file provided by "pinocchio" with
  any of the following names:

    pinocchioConfig.cmake
    pinocchio-config.cmake

  Add the installation prefix of "pinocchio" to CMAKE_PREFIX_PATH or set
  "pinocchio_DIR" to a directory containing one of the above files.  If
  "pinocchio" provides a separate development package or SDK, be sure it has
  been installed.
Call Stack (most recent call first):
  CMakeLists.txt:52 (find_package)

-- Configuring incomplete, errors occurred!
```
I didn't find the right way to install pinocchio on windows. Has anyone else experienced this problem recently?

Any help you can provide would be greatly appreciated!

BarisYazici, 2024-09-26:
> I've tried to follow the installation steps on windows (https://www.franka.cn/FCI/installation_windows.html) and used the Developer Command Prompt。But was told with following code
>
> ```
> cmake [...]

Please use the libfranka version 0.13.3 https://github.com/frankaemika/libfranka/releases/tag/0.13.3. The robot image version 5.7.0 hasn't been released yet.