frankarobotics/libfranka #89: compile libfranka on window.

THELUOHAO, 2021-01-08:
I have compiled the libfranka with CMake successfully. But for my project, I need to combine the libfranka and the frame of MFC project. I tried to compile the libfranka with the compiler of VS2017, but it threw the follow error
[image]

THELUOHAO, 2021-01-08:
Severity	Code	Description	Project	File	Line	Suppression State
Error	LNK2019	unresolved external symbol "public: __thiscall franka::JointPositions::JointPositions(class std::array<double,7> const &)" (??0JointPositions@franka@@QAE@ABV?$array@N$06@std@@@Z) referenced in function "public: class franka::JointPositions __thiscall MotionGenerator::operator()(struct franka::RobotState const &,class franka::Duration)" (??RMotionGenerator@@QAE?AVJointPositions@franka@@ABURobotState@2@VDuration@2@@Z)	myRobot2	E:\MyvsProject\myMFCrobot\myRobot2\myRobot2.obj	1

AndreasKuhner, 2024-09-26:
Hi @THELUOHAO ,
I hope you could resolve the problem!

Cheers,
Andreas

closed 2024-09-26 by AndreasKuhner