frankarobotics/libfranka #98: where is the gravity compensation term in impedance control

RanCao2018, 2021-04-09:
Hi, thank you for your work!
I have seen the cartesian impedance controller in "cartesian_impedance_control.cpp". But I found the torque command only supplies a Coriolis torque compensation (at line 105).  So where is the gravity compensation term?

peasant98, 2021-04-30:
I believe that it's taken care of under the hood.

alisonjpp, 2021-05-08:
franka does the gravity compensation itself

AndreasKuhner, 2024-09-26:
Hi @RanCao2018 ,
@alisonjpp perfectly answered it. 

For further input... the https://github.com/frankaemika/libfranka/blob/main/examples/communication_test.cpp example shows e.g. how to control the robot in 'gravity compensation mode' aka 'just send 0 torques to all joints'. Everything else is taken care of in the background.

Cheers,
Andreas

closed 2024-09-26 by AndreasKuhner