frankarobotics/libfranka #102: CMake error during build in Windows 10 + Visual Studio 2019 + vcpkg

marcofer, 2021-06-19:
I’m trying to build the library for Windows, according to [this documentation](https://frankaemika.github.io/docs/installation_windows.html), through Visual Studio 2019.

I successfully built `Eigen3 `and `Poco` through `vcpkg`. However, when I try to build libfranka I can’t understand which paths of the builds should be set to `Poco_DIR` and `Eigen3_DIR`, since I have no paths matching the suggested ones “[…]/lib/cmake/Poco” or “[…]/lib/cmake/eigen3”.

Still, independently from which paths I set, when I run the generation of the Cmake Cache in Visual Studio, I get the following error:

> [CMake] CMake Error at C:/Users/alien/Software/vcpkg/scripts/buildsystems/vcpkg.cmake:686 (set):
> 1> [CMake] Maximum recursion depth of 1000 exceeded
> 1> [CMake] Call Stack (most recent call first):
> 1> [CMake] C:/Users/alien/Software/vcpkg/scripts/buildsystems/vcpkg.cmake:764 (_find_package)

after which the last line is repeated likely for 1000 times, before aborting the process.
Any ideas to catch the reason of this error? Please tell me if I didn’t provide enough information and I’ll eventually report further build output messages. 

I apologize in advance if I'm posting this here, but there is not other support anywhere, I also tried on Franka world community without success so far.

marcofer, 2021-06-21:
Apparently I have been able to localize the source of the build error at least to avoid it to occur.
Specifically, the error above is raised when trying to build the test files in the corresponding subdirectory. In fact, setting

`option(BUILD_TESTS "Build tests" OFF)
`
does not generate any issues and the build is completed successfully. I don't know if this may be of help for you or if it's still due to some mistakes from my side, but I thought it could be worthy to be mentioned.

gaobin4216, 2021-10-30:
thank you，your comment has huge help for me

avadhutmali999, 2022-07-27:
Thankyou buddy , your troubleshooting guide helped to solve the errors in Visual studio during .json generation but what can be done to find path to the poco directory, even after numerous attempts I am unable to find them => "C:\Path\To\Poco-Install\lib\cmake\Poco". Please pass your views on this.

avadhutmali999, 2022-09-22:
> Thankyou buddy , your troubleshooting guide helped to solve the errors in Visual studio during .json generation but what can be done to find path to the poco directory, even after numerous attempts I am unable to find them => "C:\Path\To\Poco-Install\lib\cmake\Poco". Please pass your views on this.

Apparently administrator mode must be turned on before installing vcpkg, eigen3 and poco
 (i.e running cmd prompt as an administrator), else these packages wont install properly. This tip is useful to those people who are trying to building using a device of organization like university or an industry where a user account is different and has no admin rights.
In my opinion personal devices won't require admin rights but its better to enable in order to avoid getting stuck on issues.

Pat1177, 2023-03-30:
Hi, I'm currently trying to program Panda with Windows, and I'm stuck on the issue where I'm unable to find the correct path "C:\Path\To\Poco-Install\lib\cmake\Poco" in Visual Studio. I cannot find the right paths in the file system. I also set option(BUILD_TESTS "Build tests" OFF) in the CMakeLists.txt. I installed everything with administrator rights during installation of vcpkg, eigen3 and poco. Did you manage to set the paths correctly? Can you maybe help me with another advice how you could manage to get it running?

avadhutmali999, 2023-03-31:
If you have visual studio 2015 or later installed, follow these steps.

1.Make sure you have stored libfranka in C: or in C: subsequent folders.
2.You have set the build test flag to off in cmakelist.txt as mentioned above.
3.make a new folder inside libfranka e.g build, open the folder and copy the path from title bar.
4.In search type native and open the x64 prompt of visual studio in admin mode.

[image]

5. go to build directory by command cd<space>paste copied path.
6. type command 'vcpkg<space>integrate<space>install' and hit enter.
7. An output of cmake toolchain file will be shown on screen, copy it.
 
[image]

8.type "cmake<space>" paste the copied toolchain file path as shown, add the necessary flags as shown.

[image]

9. Hit enter which will starts the procedure of making Ninja files.
10. when the writing build files is completed type ninja and hit enter.

libfranka will be built.

divishadL, 2024-05-23:
@avadhutmali999 I have followed everything but this isuue still persits. What could be the problem. I am getting following error. "CMake Error at C:/Program Files/Microsoft Visual Studio/2022/Community/Common7/IDE/CommonExtensions/Microsoft/CMake/CMake/share/cmake-3.28/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find Poco (missing: Poco_INCLUDE_DIR Poco_LIBRARIES Net
  Foundation)" Same happens with eigen3 also if I just call find_package eigen3 before poco.

avadhutmali999, 2024-05-28:
@divishadL are you using a personal computer or an organization's device?
Some times many activities are blocked by organization due to security reasons including internet connections to specific sites.
Try using some less secure computer such as a personal laptop or PC.
This step doesn't require connecting robot to the system so you can still check it.