frankarobotics/libfranka #103: Expose private headers for research

ntorresalberto, 2021-07-07:
Hi,
We're currently working in a project related to [rtt-franka-hardware-integration](https://github.com/smithjoshua001/rtt-franka-hardware-integration/), which uses some private header files like `robot_control.h` in order to access `franka::RobotControl` class [here](https://github.com/smithjoshua001/rtt-franka-hardware-integration/blob/d34e97502775ab04b4f5b2967fa623cf0f50ec89/include/kinematic_chain.h#L11).

Which leads to some [hacky mechanisms](https://github.com/smithjoshua001/rtt-franka-hardware-integration/blob/d34e97502775ab04b4f5b2967fa623cf0f50ec89/CMakeLists.txt#L88) in order to include these private headers.

I don't know if this is unexpected usage, but you already seem to have a `research_interface::` namespace that could be used for this too.

I think it would be great to formalize the exposition of these headers either as a `franka/research/franka_control.h` kind in the same package or as a separate `libfranka-research-dev` sort of way. The first one seems easier to setup. 

Basically I'm trying to figure out if this is something you would be open to and if so, how to implement it. I recently worked on the [rpm packages](https://github.com/frankaemika/libfranka/pull/86) so if there's interest I could try to tackle it when I get some time.

AndreasKuhner, 2024-09-26:
Hey @ntorresalberto ,
this might be a bit late... but is there still something going on..? Maybe we could consider something for future releases :grinning: 

Cheers,
Andreas

closed 2024-09-26 by AndreasKuhner

ntorresalberto, 2024-09-26:
I'm no longer involved in the project but by the looks of it, `franka::RobotControl` seems to still be a virtual interface defined in a private header ([src/robot_control.h](https://github.com/frankaemika/libfranka/blob/main/src/robot_control.h)).

It's old code now, so I don't know whether this is the only/best way to achieve it but, at that point in time, they needed a `std::unique_ptr < franka::RobotControl > franka_control` in order to use different `franka::Robot::Impl` objects (see [here](https://github.com/smithjoshua001/rtt-franka-hardware-integration/blob/d34e97502775ab04b4f5b2967fa623cf0f50ec89/src/kinematic_chain.cpp#L16)). 

The proposal from this issue was to formalize that use case by exposing `franka::RobotControl` into the `research_interface::` interface. However, as I'm no longer working on it and so I can't tell if it'd be useful anymore.