frankarobotics/libfranka #109: Meaning of Elbow

peetCreative, 2021-12-08:
Hey,
This might be up to a lack of understanding for robotarms  in general or of the Franka Emika in particular.
As I work with FCI I keep stumbeling upon the elbow. It is given by robot_state and can be sepperatly commanded in cartesian motion generation. I wonder what it is there for. What is the purpose additionally being able to configure Position of the 3rd joint in [rad] and Sign of the 4th joint? It just looks random to me. Why can I only change the sign of the 4th joint and not the whole position? What would it be anyway good for?
I suspect I'm just missing some insights of robotics, or it would be nice to have a bit more documentation.
Thanks!

marcbone, 2021-12-09:
You have 6 DoF in cartesian space (x, y, z, roll, pitch, yaw). However, the robot has 7 DoF. Therefore, you can freely choose the position of one joint, which is the 3rd in our case. There are then 2 solutions for the 4th joint. However, in general the best idea is to set the desired sign of the 4th joint to the current one.

If you want, you can run the "generate_elbow_motion" example to see the effect of the elbow value.

peetCreative, 2021-12-15:
OK, I think than it's just that. I think I got it.

closed 2021-12-15 by peetCreative