frankarobotics/libfranka #110: Change Configured Torque Thresholds in Simulink

JBLip, 2021-12-19:
So in our project, we are trying to control the panda robot to interact with a human touch. The thing is, sometimes we get the following error: 
[image]
On the Desk, I think I found where I can change the torque thresholds,
[image]
 but I don't seem to be able to find how to do this in Simulink. Changing the collision-behavior array (52x1) doesn't seem to make a difference. 

Can anyone please help me?

marcbone, 2021-12-20:
I dont know what you are exactly using but if you use libfranka you can try to set the collision behavior to 100 for everything before you start the control loop.

```cpp
    robot.setCollisionBehavior({{100.0, 100.0, 100.0, 100.0, 100.0, 100.0, 100.0}},
                               {{100.0, 100.0, 100.0, 100.0, 100.0, 100.0, 100.0}},
                               {{100.0, 100.0, 100.0, 100.0, 100.0, 100.0}},
                               {{100.0, 100.0, 100.0, 100.0, 100.0, 100.0}});
```

If you are then still hitting the limits you are pushing it too hard, or there is something wrong with your controller.
To get a feel how much you can push the robot the robot you can try to run the `cartesian_impedance_control` example from libfranka.

AndreasKuhner, 2024-09-26:
Hi @JBLip ,
I hope the issue is resolved. If not, don't hesitate to open the issue again.

Cheers,
Andreas

closed 2024-09-26 by AndreasKuhner