frankarobotics/libfranka #111: Stop gripper dont work sometimes

lmmartine, 2021-12-21:
(on the gripper) I am trying to use the method stop to stop the move method of the gripper. However almost half of the times dont work.

marcbone, 2021-12-21:
We are aware of the problem that the stop method does not work reliably. Unfortunately, this bug is very time consuming to fix and does not have a high priority from our side. So dont expect any progress on this soon. Could you tell us more about your use-case?

lmmartine, 2021-12-21:
We are using tactile sensors in the fingers and we want to stop the gripper after the sensors detect something. Is there any other way to doing it?

marcbone, 2021-12-22:
I like that use-case. Since, you cannot rely on the stop method to stop the movement, there is sadly no way to use your tactile sensor to control the gripper. However, you can try to use the grasp method and use very high epsilon values. That way the gripper will stop as soon as it detects something, basically eliminating the need for your tactile sensor (however, your sensor is probably much more precise). Keep in mind that the gripper will then still apply a force of at least 30 Newton (this is another problem with the gripper) to the object. You can call the stop method immediately afterwards to release the pressure onto the object (this is actually mainly intended use of the stop method). This should work for rigid objects. However, you probably cant grasp an egg this way :)

knazari, 2022-01-06:
I am also trying to command a grasp force with the grasp() method by giving it large epsilon values. But it still applies a large grip force even though the given force argument is small. @marcbone Is there a way that we could change this minimum 30N grasp force in libfranka? 30N minimum grasp force is too large for the objects within robot's max load range.

AndreasKuhner, 2025-03-05:
If there are still some issues, please let us know

closed 2025-03-05 by AndreasKuhner