frankarobotics/libfranka #117: How to restart after button pushed

arnauddomes, 2022-05-04:
Hello,

I want to be able to use the push button for the guide mode in moveit as a way to retrieve position data from robot arm.

When moveit launch file is start, I push the button. Then I still can't retrieve arm position but after I unlock the button, I can't send anymore the trajectory to moveit.

Is there a way to allow it by restarting something on libfranka?

avadhutmali999, 2022-08-02:
A possible way could be positioning the robot in the required pose, pressing down the activation button, retrieving the position data. Again resetting the external activation key to default state and then again moving the robot to the next required position. 
Although its a tedious and repetitive task, its the only option currently as per my knowledge.

AndreasKuhner, 2024-09-26:
Hi @arnauddomes ,
if you still have some issues, we are happy to hear about it!

For your question, I guess it would be unsafe to have some automatic motion being active while someone could be around the robot/close to the guiding button... :)

Cheers,
Andreas

closed 2024-09-26 by AndreasKuhner