frankarobotics/libfranka #122: Are there any plan to support Ubuntu 22.04

CWEzio, 2022-08-03:
As ubuntu 22.04 is officially released, are there any plan for libfranka to support it? Are there any existing compatibility issues with ubuntu 22.04 in current version libfranka?

marcbone, 2022-08-03:
Hello, we never tested libfranka on Ubuntu 22.04. We only test it with Ubuntu 20.04. However, I see no reason why it should not work with Ubuntu 22.04 or any other linux distro. There shouldnt be any problems. Feel free to open a new issue if you are having any problems to get it to run with Ubuntu 22.04

closed 2022-08-03 by marcbone

yijionglin, 2023-10-25:
Hi @marcbone ,

Hope you are well!

I tried to use libfranka in ubuntu 22, and as my system version is 4, I used libfranka version 0.8.0. However I met this problem when I did the cmake --build . :

 error: ‘string’ is not a member of ‘std’  74 | bool setCurrentThreadToHighestSchedulerPriority(std::string* error_message);

Wondering if you have any ideas on this bit?

Best,
Bourne

yijionglin, 2023-10-27:
> Hi @marcbone ,
>
> Hope you are well!
>
> I tried to use libfranka in ubuntu 22, and as my system version is 4, I used libfranka version 0.8.0. However I met this problem when I did the cmake --build . : [...]

I found [this](https://github.com/frankaemika/libfranka/issues/65) solved my problem.