frankarobotics/libfranka #130: Incompatible Version Exception when trying the robot's connection

ValerioGio, 2022-11-14:
Hello there,
At the beginning, this error came out because I have installed the libfranka version 0.10 which is not compatible with system version 4.1.1, as seen in desk. So, I've downgraded libfranka to 0.8.0 as suggested from changelog. In fact, with the libfranka examples, now I am able to run echo_robot_state without problems, and get data from robot. 
However, when I tried to include the library in my own project, I get again the following error:
[image]
I am really struggling to understand where the problem could be. Can someone help, please?

marcbone, 2022-11-15:
0.8.0 is the correct libfranka version for your system. My guess is that you still have 0.10 installed on your machine and that you projects tries to use this version. You can verify this by printing the franka version in your CMakeLists.txt with:
`message(libfranka version : ${Franka_VERSION})`
You should then remove the 0.10 version.
Also you can enforce that your project is using the correct version of libfranka by using:
`find_package(Franka 0.8.0 EXACT REQUIRED)`

ValerioGio, 2022-11-15:
Thanks for your answer. It was good hint, the problem was indeed on the old franka.dll coming from the 0.10.0 version that needed to be manually replaced

closed 2022-11-15 by ValerioGio