frankarobotics/libfranka #131: "communication_constraints_violation" trying to run franka examples on windows 11

ValerioGio, 2022-11-15:
Hello there,
I've the following error when I try to launch the various examples in franka's examples folder:

[image]

I've already tried all the suggestions on https://frankaemika.github.io/docs/troubleshooting.html except for  CPU frequency scaling because the instruction is for Linux only. Anyway, the CPU base velocity is 2.4 GHz. Also I've tried to modify the energy setup in order to improve performance but the result is the same.
Any idea please?

marcbone, 2022-11-16:
This is why we do not recommend to use Windows. Windows is not real-time capable so a certain amount of communication constraints violations can be expected. Make sure that you compile everything in Release mode, close all other programs and maybe try out a USB 3.0 to Ethernet dongle. If this does not help, I am afraid we are unable to help you, but if you find some windows specific tweak to make it work, please report it back to us.

ValerioGio, 2022-11-17:
Thanks for the fast reply Marco.
I have done various tests using another PC, always in windows, and the problem is the ethernet port: if I use an ethernet adapter, neither type-C or USB 3.0, the error appears and the robot doesn't move. Instead, using the pc with the direct ethernet port, the robot moves without problem, even if the "communication_constraints_violation" still appears.

marcbone, 2022-11-21:
You are welcome. However, I dont get what you mean with 

>  the robot moves without problem, even if the "communication_constraints_violation" still appears

You cannot move the robot while if you are getting this error. It will make your controller crash. After that you have to either use the automaticErrorRecovery method or toggle the user stop button.

AndreasKuhner, 2024-09-26:
Hi @ValerioGio ,
I close this issue - if you have any further problems, please let us know.

Cheers,
Andreas

closed 2024-09-26 by AndreasKuhner