frankarobotics/libfranka #133: Unable to run some libfranka examples [libfranka: Move command rejected: command not possible in the current mode]

juanjqo, 2023-02-10:
Hi guys,

I am having a hard time trying to run some libfranka examples on our new Franka Research 3. I am using a desktop PC with Ubuntu 20.04 LTS (5.9.1-rt20) + FCI (5.2.2). I followed the steps of the [official documentation](https://frankaemika.github.io/docs/index.html). I built from sources the libfranka library. Furthermore, I am using both the external enabling device and the emergency stop device provided by Franka Emika.

[image]

Once I enable the FCI via Desk and release the emergency stop device, the status lights on both sides of the base turn on in blue.

[image]

```shell
moonshot@moonshot:~/Documents/git/libfranka/build/examples$ ./communication_test 172.16.0.2
WARNING: This example will move the robot! Please make sure to have the user stop button at hand!
Press Enter to continue...

libfranka: Move command rejected: command not possible in the current mode ("User stopped")!
```
However, the example echo_robot_state works:

```shell
moonshot@moonshot:~/Documents/git/libfranka/build/examples$ ./echo_robot_state 172.16.0.2
{"O_T_EE": [0.999232,-0.0119429,-0.0370718,0,-0.0370204,0.00449629,-0.999295,0,0.0121014,0.999919,0.00405078,0,0.243673,0.207435,0.267699,1], "O_T_EE_d": [0.999232,-0.0119468,-0.0370653,0,-0.0370139,0.00450112,-0.999295,0,0.0121054,0.999919,0.00405555,0,0.243676,0.207435,0.267702,1], "F_T_NE": [0.7071,-0.7071,0,0,0.7071,0.7071,0,0,0,0,1,0,0,0,0.1034,1], "NE_T_EE": [1,0,0,0,0,1,0,0,0,0,1,0,0,0,0,1], "F_T_EE": [0.7071,-0.7071,0,0,0.7071,0.7071,0,0,0,0,1,0,0,0,0.1034,1], "EE_T_K": [1,0,0,0,0,1,0,0,0,0,1,0,0,0,0,1], "m_ee": 0.73, "F_x_Cee": [-0.01,0,0.03], "I_ee": [0.001,0,0,0,0.0025,0,0,0,0.0017], "m_load": 0, "F_x_Cload": [0,0,0], "I_load": [0.01,0,0,0,0.01,0,0,0,0.01], "m_total": 0.73, "F_x_Ctotal": [-0.01,0,0.03], "I_total": [0.001,0,0,0,0.0025,0,0,0,0.0017], "elbow": [-0.00288152,-1], "elbow_d": [-0.00288453,-1], "elbow_c": [-0.00288453,-1], "delbow_c": [0,0], "ddelbow_c": [0,0], "tau_J": [-0.111287,-10.4038,-0.539959,15.4589,0.560318,-1.37382,0.0223021], "tau_J_d": [0,0,0,0,0,0,0], "dtau_J": [-13.0128,-2.11975,-82.3101,35.3887,-47.8911,7.69937,-3.09992], "q": [-0.011071,-0.36645,-0.00288152,-3.06383,1.5706,1.56736,-0.304106], "dq": [-9.89951e-05,-1.10821e-06,0.00119321,-0.000213415,-0.000403995,0.000216784,0.000423343], "q_d": [-0.0110693,-0.366452,-0.00288453,-3.06383,1.5706,1.56736,-0.304107], "dq_d": [0,0,0,0,0,0,0], "ddq_d": [0,0,0,0,0,0,0], "joint_contact": [0,0,0,0,0,0,0], "cartesian_contact": [0,0,0,0,0,0], "joint_collision": [0,0,0,0,0,0,0], "cartesian_collision": [0,0,0,0,0,0], "tau_ext_hat_filtered": [-0.13086,-0.0247789,0.11404,-0.0615543,-0.133433,0.0775263,-0.0808499], "O_F_ext_hat_K": [-0.257623,-0.606792,-0.164557,0.0482181,-0.108973,-0.0392122], "K_F_ext_hat_K": [-0.244083,0.171254,-0.610526,-0.0811111,-0.0525614,-0.0808452], "O_dP_EE_d": [0,0,0,0,0,0], "O_ddP_O": [0,0,-9.81], "O_T_EE_c": [0.999232,-0.0119468,-0.0370653,0,-0.0370139,0.00450112,-0.999295,0,0.0121054,0.999919,0.00405555,0,0.243676,0.207435,0.267702,1], "O_dP_EE_c": [0,0,0,0,0,0], "O_ddP_EE_c": [0,0,0,0,0,0], "theta": [-0.0106292,-0.366689,-0.00264612,-3.06514,1.56928,1.56687,-0.30881], "dtheta": [0,0,0,0,0,0,0], "current_errors": [], "last_motion_errors": [], "control_command_success_rate": 0, "robot_mode": "User stopped", "time": 121891}
```

According to this [discussion](https://www.franka-community.de/t/libfranka-move-command-rejected-command-not-possible-in-the-current-mode-user-stopped/4454), FR3 has enabled safety rules by default. I checked the Watchman settings

[image]

There are safety rules related to X3.2 and X3.3 inputs. I deleted them to test the examples. 

[image]

I tested again, the example did not work.

```shell
moonshot@moonshot:~/Documents/git/libfranka/build/examples$ ./communication_test 172.16.0.2
WARNING: This example will move the robot! Please make sure to have the user stop button at hand!
Press Enter to continue...

libfranka: Move command rejected: command not possible in the current mode ("Other")! Did you open the brakes?
```

The example echo_robot_state still works:

```shell
moonshot@moonshot:~/Documents/git/libfranka/build/examples$ ./echo_robot_state 172.16.0.2
{"O_T_EE": [0.999231,-0.0119636,-0.03707,0,-0.0370185,0.00450183,-0.999295,0,0.0121223,0.999918,0.00405558,0,0.243691,0.207438,0.267708,1], "O_T_EE_d": [0.999231,-0.0119662,-0.0370701,0,-0.0370186,0.00449949,-0.999295,0,0.0121248,0.999918,0.00405314,0,0.24369,0.207437,0.267707,1], "F_T_NE": [0.7071,-0.7071,0,0,0.7071,0.7071,0,0,0,0,1,0,0,0,0.1034,1], "NE_T_EE": [1,0,0,0,0,1,0,0,0,0,1,0,0,0,0,1], "F_T_EE": [0.7071,-0.7071,0,0,0.7071,0.7071,0,0,0,0,1,0,0,0,0.1034,1], "EE_T_K": [1,0,0,0,0,1,0,0,0,0,1,0,0,0,0,1], "m_ee": 0.73, "F_x_Cee": [-0.01,0,0.03], "I_ee": [0.001,0,0,0,0.0025,0,0,0,0.0017], "m_load": 0, "F_x_Cload": [0,0,0], "I_load": [1,0,0,0,1,0,0,0,1], "m_total": 0.73, "F_x_Ctotal": [-0.01,0,0.03], "I_total": [0.001,0,0,0,0.0025,0,0,0,0.0017], "elbow": [-0.00288991,-1], "elbow_d": [-0.00288991,-1], "elbow_c": [-0.00288991,-1], "delbow_c": [0,0], "ddelbow_c": [0,0], "tau_J": [-0.162862,-10.4038,-0.484417,15.4113,0.574051,-1.361,0.00856917], "tau_J_d": [0,0,0,0,0,0,0], "dtau_J": [-81.7887,-58.9163,-27.4926,-34.5199,-7.6503,13.2805,-38.7108], "q": [-0.0110475,-0.366439,-0.00288991,-3.0638,1.57063,1.56737,-0.304082], "dq": [0.000860098,0.000565486,-0.000791949,0.00106009,-5.67854e-05,-0.00161667,0.00140955], "q_d": [-0.0110496,-0.366441,-0.00288991,-3.0638,1.57063,1.56737,-0.304084], "dq_d": [0,0,0,0,0,0,0], "ddq_d": [0,0,0,0,0,0,0], "joint_contact": [0,0,0,0,0,0,0], "cartesian_contact": [0,0,0,0,0,0], "joint_collision": [0,0,0,0,0,0,0], "cartesian_collision": [0,0,0,0,0,0], "tau_ext_hat_filtered": [-0.128403,-0.0322772,0.11902,-0.0811255,-0.129441,0.0773063,-0.0791694], "O_F_ext_hat_K": [-0.347445,-0.568577,-0.101644,0.0362465,-0.146392,-0.0352466], "K_F_ext_hat_K": [-0.336614,0.111877,-0.573154,-0.0950299,-0.0280528,-0.0791656], "O_dP_EE_d": [0,0,0,0,0,0], "O_ddP_O": [0,0,-9.81], "O_T_EE_c": [0.999231,-0.0119662,-0.0370701,0,-0.0370186,0.00449949,-0.999295,0,0.0121248,0.999918,0.00405314,0,0.24369,0.207437,0.267707,1], "O_dP_EE_c": [0,0,0,0,0,0], "O_ddP_EE_c": [0,0,0,0,0,0], "theta": [-0.0106292,-0.366682,-0.00264612,-3.06513,1.56929,1.56686,-0.308819], "dtheta": [0,0,0,0,0,0,0], "current_errors": [], "last_motion_errors": [], "control_command_success_rate": 0, "robot_mode": "Other", "time": 738619}
```
I would appreciate if you could help me or guide me to a solution. Thank you in advance.

Best regards, 

Juan

marcbone, 2023-02-10:
Just to be sure: Did you open the brakes? :sweat_smile:

juanjqo, 2023-02-10:
@marcbone thank you for your reply. I did not know how to open the brakes. I just realized that the padlock icon is to open/close the brakes. Now the examples are working.

Thank you for your time!

closed 2023-02-10 by juanjqo

marcbone, 2023-02-10:
Glad I could help. Thanks for putting so much effort into the description of your problem. Was the best I have seen here for a long time and this makes it much easier for me to solve your problem.