frankarobotics/libfranka #135: Missing headers in 0.7.1 and 0.8.0

JeanElsner, 2023-03-06:
For some legacy systems we still need to build libfranka versions 0.7.1 and 0.8.0. However, these releases no longer build out of the box with GCC. Specifically, there are two headers missing:
https://github.com/frankaemika/libfranka/blob/80197c6f57d87e4a4d0fe298e89107ab46543808/include/franka/control_tools.h#L6
is missing \<string\> and 
https://github.com/frankaemika/libfranka/blob/80197c6f57d87e4a4d0fe298e89107ab46543808/src/control_types.cpp#L3
is missing \<stdexcept\>.
Maybe the include order changed with recent GCC versions or something and while it's an easy fix it prevents our CI from directly checking out those releases and building them (tested with Ubuntu 22.04 and manylinux2014). If you could bump those releases it would help me streamline the pipeline a bit, which would be greatly appreciated :)

andrejpan, 2023-03-07:
Fixing old releases on newer operating systems is currently impossible due to resource limitations. Please either upgrade your Franka Emika robot or perhaps use a docker image with an older Ubuntu version.

drewskoots, 2023-05-19:
@JeanElsner I was able to build this in Docker using sed to make the 2 required edits. Maybe that could be of help for you as well.

AndreasKuhner, 2024-09-26:
Hi @drewskoots ,
thank you for the workaround, I hope that helps @JeanElsner .

Cheers,
Andreas

closed 2024-09-26 by AndreasKuhner