frankarobotics/libfranka #137: Motion finished commanded, but the robot is still moving!

yetian0, 2023-05-18:
Hi, I met this question "Motion finished commanded, but the robot is still moving! " between franka finished one movement and restart for another movement.  I can not even find which kind of exception this is, can anyone help with it? Thanks.

marcbone, 2023-05-19:
This should be a franka::ControlException. Usually this happens when you commanded too high velocities at some point in your trajectory with the rate limiter enabled. The rate limiter will limit your commanded velocity, but when you are now send your last command the robot is still moving since it tries to catch up. To fix this you should disable the rate_limiter in the control method:
```cpp
robot.control(your_control_callback_function, false);
```
When you use an FR3 this is probably already enough to make it work. If you are using a Panda you probably also have to make your trajectory slower.

closed 2023-07-20 by marcbone

104H, 2023-12-01:
Hi. I am using the Franka Production 3 and I face this error. I am using the [JointPosition based controller](https://frankaemika.github.io/libfranka/classfranka_1_1Robot.html#a8a41a2f208cc0c7e68569a550af4e1b2).

The rate limiter is false by default. The cutoff frequency is set to 100.0 (by default).

The exactly error statement is:
```
Motion finished commanded, but the robot is still moving! ["joint_motion_generator_velocity_discontinuity", "joint_motion_generator_acceleration_discontinuity"]
```