frankarobotics/libfranka #140: Opening Brakes Failed

homangab, 2023-08-20:
Hi, When I click on "Unlock Joints" from the Franka dashboard (robot is yellow), it displays an error message "Opening Brakes Failed" and does not turn blue - this happened suddenly, and the robot has been working fine prior to this. Could you kindly advise on potential fixes? (i tried turning off and on again several times, and also made sure the stop or brakes are non on)

Thanks!
-- Homanga

[image]

davidschnermann, 2024-03-15:
Hi Homanga,

we have the same problem now. How did you fixed it?

Thanks.

David

AndreasKuhner, 2024-11-08:
Hi @homangab and @davidschnermann ,
I guess for this error, the best contact point is [support@franka.de](mailto:support@franka.de).

If anything further is open, don't hesitate to open a new issue :robot:

closed 2024-11-08 by AndreasKuhner