frankarobotics/libfranka #141: libfranka 0.11.0

saivojjala, 2023-08-27:
Hi, I was recently attempting to build franka_ros2 for ros2 humble. However the package franka_hardware requires libfranka version 0.11.0. I tried using a different version 0.9.2 and 0.10.0 but the build threw an error as those version seem to be missing the active_control.h header file. 

Please look into this, I have raised an issue in franka ros2 repository as well, but no response yet.

BarisYazici, 2023-08-27:
For latest franka_ros2 [V0.1.3](https://github.com/frankaemika/franka_ros2/releases/tag/v0.1.3) you need the fr3-develop branch from the libfranka repository which is the v0.11.0.

BarisYazici, 2023-08-27:
You can use an earlier franka_ros2 version v0.1.0 which works with libfranka v0.10.0

saivojjala, 2023-08-27:
okay thank you, it built successfuly now

closed 2023-08-27 by saivojjala