frankarobotics/libfranka #143: Robots keep getting "power_limit_violation" error.

trannguyenle95, 2023-09-25:
Hi @frankaemika,

I have a question regarding the "power limit violation". We tried to run our custom-made controller for the robot to follow a trajectory at a pretty high speed. Sometimes it works, however, we often get the "power_limit_violation" error.

We checked the docs and it says "If the maximum allowed power is reached, the power_limit_violation will be triggered. It will prevent the robot from moving and continuing the control loop." Can you elaborate on what is "maximum allowed power" here? What could be the cause of this problem? Can you provide any hints for solving this?

P/s: The error happens on both Franka 2 and 3 but mostly Franka 2.

chaitanyantr, 2024-01-23:
Hi @trannguyenle95 ,

Did you figure out any solution for this? I'm getting "power_limit_violation" as well.

**setup:-**
Franka panda.
ubuntu 22
ROS2
Python
FCI Active mode

trannguyenle95, 2024-01-23:
Not really, in the end, I had to execute my trajectory in a slower manner.

AndreasKuhner, 2024-09-26:
Hi @trannguyenle95 ,
the background is that due the dimensioning of the robot, it sometimes is not possible to move/accelerate with all joints full speed at once. Depends of course mostly on the needed torque per joint (the more torque a joints needs to use, the more power it consumes). And at the end of the day, all joints take from the same power source (which is limited). Trajectories close to the center need less torque for the same movements compared to far away trajectories (e.g. outstreched arm). 

As you found out, reducing the acceleration/speed reduces/resolves the problem :)

Cheers,
Andreas

closed 2024-11-08 by AndreasKuhner

younghyopark, 2025-08-31:
Hi @AndreasKuhner , 

Can you specify what the exactly is the limit (and its unit) that triggers this `power_limit_violation` error? I tried to search for this value in every available documents for FR3, but I was not able to find that number.

AndreasKuhner, 2025-09-03:
The max limit is 376 watt while the soft limit is 276 watt. Meaning, if the robot surpasses the 276 for a 'while', we also throw an error. Hence, we can shortly deal with peaks up to 376 but not for a long time.