frankarobotics/libfranka #144: Libfranka cmake error

Jacklove-G, 2023-10-11:
My Os is 4.2.2, and I downloaded libfranka-0.9.1 and libfranka-0.9.2,
when I run cmake -DCMAKE_BUILD_TYPE=Release -DBUILD_TESTS=OFF ..

for  libfranka-0.9.1, there is 
CMake Error at cmake/SetVersionFromGit.cmake:26 (message):
  /home/g/libfranka does not seem to be a git repository, cannot find a
  '.git/' folder inside it?
Call Stack (most recent call first):
  CMakeLists.txt:5 (set_version_from_git)

-- Configuring incomplete, errors occurred!

For  libfranka-0.9.2, there is 
-- Found Threads: TRUE  
CMake Error at CMakeLists.txt:53 (add_subdirectory):
  The source directory

    /home/g/libfranka/common

  does not contain a CMakeLists.txt file.

-- Configuring incomplete, errors occurred!
See also "/home/g/libfranka/build/CMakeFiles/CMakeOutput.log".
See also "/home/g/libfranka/build/CMakeFiles/CMakeError.log".

What should I do?
And I use Ubuntu18.04 melodic.

christian-rauch, 2023-11-10:
Did you clone the repo with `--recursive` as suggested here: https://frankaemika.github.io/docs/installation_linux.html#building-libfranka?

AndreasKuhner, 2024-09-26:
Hey @Jacklove-G ,
did this resolve the issue?

Cheers,
Andreas

closed 2024-11-08 by AndreasKuhner